ros2/ros1_bridge #248: ros1_bridge build failure

McKayCyborg, 2020-03-16:
My system:
ROS1 Melodic, ROS2 Dashing on an ARM64 RPi4 running Ubuntu 18.04 Bionic

Issue overview:
I am trying to build ros1_bridge according to the instructions outlined here: https://dev.px4.io/master/en/middleware/micrortps.html (see section _Building the workspaces_ steps 6 and 7). The step that fails is simply building the ros1_bridge package using colcon build. I initially posted this in the px4_ros_com github and they have referred me here as they think it is an issue with ros1_bridge.

Steps to reproduce:

1. sudo apt install ros-dashing-ros-base
2. `mkdir -p ~/px4_ros_com_ros2/src`
3. `git clone https://github.com/PX4/px4_ros_com.git ~/px4_ros_com_ros2/src/px4_ros_com`
4. `git clone https://github.com/PX4/px4_msgs.git ~/px4_ros_com_ros2/src/px4_msgs`
5. `mkdir -p ~/px4_ros_com_ros1/src`
6. `git clone https://github.com/PX4/px4_ros_com.git ~/px4_ros_com_ros1/src/px4_ros_com -b ros1`
7. `git clone https://github.com/PX4/px4_msgs.git ~/px4_ros_com_ros1/src/px4_msgs -b ros1`
8. `source /opt/ros/dashing/setup.bash`
9. `git clone https://github.com/ros2/ros1_bridge.git -b dashing ~/px4_ros_com_ros2/src/ros1_bridge`
10. `colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+` - this build succeeds without issues.
11. Install ROS1 using sudo apt install ros-melodic-ros-base
12. New terminal
13. `source /opt/ros/melodic/setup.bash`
14. `cd ~/px4_ros_com_ros1 && colcon build --symlink-install --event-handlers console_direct+` - this build succeeds.
15. `source ~/px4_ros_com_ros1/install/setup.bash && source ~/px4_ros_com_ros2/install/setup.bash`
16. `cd ~/px4_ros_com_ros2 && colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+`

Expected behaviour:
Successful build of ros1_bridge

Actual behaviour:
Failure during building of simple_bridge_2_to_1.cpp.o with the following error:

`Traceback (most recent call last): File "bin/ros1_bridge_generate_factories", line 11, in <module> from ros1_bridge import generate_cpp File "/home/ubuntu/px4_ros_com_ros2/src/ros1_bridge/ros1_bridge/__init__.py", line 709, in <module> rosidl_parser.Field.__hash__ = FieldHash AttributeError: module 'rosidl_parser' has no attribute 'Field' make[2]: *** [generated/get_factory.cpp] Error 1 CMakeFiles/ros1_bridge.dir/build.make:61: recipe for target 'generated/get_factory.cpp' failed make[1]: *** [CMakeFiles/ros1_bridge.dir/all] Error 2 make[1]: *** Waiting for unfinished jobs.... CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ros1_bridge.dir/all' failed`

My comments:
Unsure if this invalid member reference is a developer issue or if it is an issue with my environment. Any help would be greatly appreciated.

dirk-thomas, 2020-03-16:
This line number and content doesn't make sense for the Dashing branch:

> `/src/ros1_bridge/ros1_bridge/__init__.py", line 709, in <module> rosidl_parser.Field.__hash__ = FieldHash AttributeError: module 'rosidl_parser' has no attribute 'Field'`

Can you please share the exact commit you are using (`cd ~/px4_ros_com_ros2/src/ros1_bridge && git log -n 1`).

McKayCyborg, 2020-03-16:
Hi Dirk, thanks for the reply. 

Using the command: `cd ~/px4_ros_com_ros2/src/ros1_bridge && git log -n 1`
Yields this output: 
`commit 6614341300fce960b2d0ea26400ed56c2d2409db (HEAD -> crystal, origin/crystal)
Author: Dirk Thomas <dirk-thomas@users.noreply.github.com>
Date:   Mon May 20 14:10:24 2019 -0700

    add missing test dependency in Crystal (#198)
    
    Signed-off-by: Dirk Thomas <dirk-thomas@users.noreply.github.com>`

It seems that the ros1_bridge branch is not the dashing branch? Is this what could be causing the issue?

dirk-thomas, 2020-03-17:
> It seems that the ros1_bridge branch is not the dashing branch? Is this what could be causing the issue?

Yes, that is very likely the cause. If you use Dashing you should use the `dashing` branch of this repository.

> `git clone https://github.com/ros2/ros1_bridge.git -b dashing ~/px4_ros_com_ros2/src/ros1_bridge`

If you have the `crystal` branch you must have diverged from the instructions posted above.

Assuming using the right branch resolves the problem I am going to close this ticket.

closed 2020-03-17 by dirk-thomas