ros2/ros1_bridge #251: ros1_bridge always lost the previous message

lx852357, 2020-04-02:
Env:
1. ros1 container(melodic)
2. ros1_bridge container(melodic-eloquent)
3. ros2 container(eloquent)

Describe:
I used dynamic_bridge
```
ros2 run ros1_bridge dynamic_bridge
```

1. ros1 publisher in 1 hz. 
First, I run the ros2 listener then run ros1 publisher.
``` ros1 container
[INFO] [1585816056.180143]: hello world 0
[INFO] [1585816057.181246]: hello world 1
[INFO] [1585816058.181275]: hello world 2
[INFO] [1585816059.181073]: hello world 3
[INFO] [1585816060.180717]: hello world 4
[INFO] [1585816061.181281]: hello world 5
[INFO] [1585816062.181247]: hello world 6
[INFO] [1585816063.181265]: hello world 7
[INFO] [1585816064.181009]: hello world 8
[INFO] [1585816065.180690]: hello world 9

```
and in ros2 container
``` ros2 container
[INFO] [minimal_subscriber]: I heard: "hello world 1"
[INFO] [minimal_subscriber]: I heard: "hello world 2"
[INFO] [minimal_subscriber]: I heard: "hello world 3"
[INFO] [minimal_subscriber]: I heard: "hello world 4"
[INFO] [minimal_subscriber]: I heard: "hello world 5"
[INFO] [minimal_subscriber]: I heard: "hello world 6"
[INFO] [minimal_subscriber]: I heard: "hello world 7"
[INFO] [minimal_subscriber]: I heard: "hello world 8"
[INFO] [minimal_subscriber]: I heard: "hello world 9"

```
the ros2 terminal lost the message `[INFO] [1585816056.180143]: hello world 0`

2. ros1 publisher in 50 hz.
``` ros1 container
[INFO] [1585816056.180143]: hello world 0
[INFO] [1585816057.181246]: hello world 1
...
[INFO] [1585816910.092495]: hello world 36
[INFO] [1585816910.112683]: hello world 37
[INFO] [1585816910.132212]: hello world 38
[INFO] [1585816910.152173]: hello world 39
[INFO] [1585816910.172155]: hello world 40
[INFO] [1585816910.192415]: hello world 41
[INFO] [1585816910.212640]: hello world 42
[INFO] [1585816910.232528]: hello world 43
[INFO] [1585816910.252540]: hello world 44
[INFO] [1585816910.272453]: hello world 45
[INFO] [1585816910.292512]: hello world 46
```
and in ros2 container
``` ros2 container
[INFO] [minimal_subscriber]: I heard: "hello world 36"
[INFO] [minimal_subscriber]: I heard: "hello world 37"
[INFO] [minimal_subscriber]: I heard: "hello world 38"
[INFO] [minimal_subscriber]: I heard: "hello world 39"
[INFO] [minimal_subscriber]: I heard: "hello world 40"
[INFO] [minimal_subscriber]: I heard: "hello world 41"
[INFO] [minimal_subscriber]: I heard: "hello world 42"
[INFO] [minimal_subscriber]: I heard: "hello world 43"
[INFO] [minimal_subscriber]: I heard: "hello world 44"
[INFO] [minimal_subscriber]: I heard: "hello world 45"
[INFO] [minimal_subscriber]: I heard: "hello world 46"
```
I lost the first 35 messages.

I have set up qos in subscriber.
```
def main(args=None):
    global g_node
    rclpy.init(args=None)
    g_node = rclpy.create_node('minimal_subscriber')

    qos_profile = QoSProfile(
            depth = 100,
            reliability = QoSReliabilityPolicy.RMW_QOS_POLICY_RELIABILITY_RELIABLE
            )

    subscription = g_node.create_subscription(String, 'topic', chatter_callback,qos_profile)
    subscription  # prevent unused variable warning

    while rclpy.ok():
        rclpy.spin_once(g_node)
    g_node.get_logger().info('The spin was ended!')
    print("what's up")
    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    g_node.destroy_node()
    rclpy.shutdown()

```
I use pure ros2 publisher and subscriber to communication and it does not lose any messages. So I think it's because of ros1_bridge.
**What can I do to not lose the previous message?**

lx852357, 2020-04-03:
I think I have found the reason but I don't know if I should report as a bug.

I thought the dynamic_bridge behavior should be that passing the message to the topic with or without subscriber. I have some tests.

1. **ros2 publisher without ros1 subscriber.** The dynamic_bridge will automatically created a subscriber. Then the callback function publish a ros1 message to topic.

2.  **ros1 publisher without ros2 subscriber.**  The dynamic_bridge will not created a subscriber to bridge message unless I run a ros2 subscriber.

In the case of **ros2 publisher without ros1 subscriber**,  when I run a ros2 publisher [ros1_subscriber == ros1_subscribers.end()](https://github.com/ros2/ros1_bridge/blob/master/src/dynamic_bridge.cpp#L216) will become to false. And in the case of **ros1 publisher without ros2 subscriber**, [ros2_subscriber == ros2_subscribers.end()](https://github.com/ros2/ros1_bridge/blob/master/src/dynamic_bridge.cpp#L152) always be true. Besides, I tried to set [bridge_all_1to2_topics](https://github.com/ros2/ros1_bridge/blob/master/src/dynamic_bridge.cpp#L153) as true. The created 1to2 bridge will be removed immediately.
[image]

fujitatomoya, 2020-04-06:
@lx852357 

can you try with the option **--bridge-all-topics** for ros1_bridge dynamic_bridge? dynamic_bridge checks if publication and subscription is actually needed on the other side or not. (for example, even if there is a ros1 publisher, it does not make publication on the ros2 side unless if nobody subscribes on that topic in ros2.)

lx852357, 2020-04-07:
I used option **--bridge-all-topics** for dynamic_bridge, but still lost the previous message. And same as I said before, The dynamic_bridge **not** created a subscriber for my ros1 publisher until I run a ros2 subscriber.

dirk-thomas, 2020-04-07:
@lx852357 both of your statements contradict each other. If you pass that option the bridge gets created right away - not waiting for subscribers to exist.

lx852357, 2020-04-07:
I'm sorry for that I used option --bridge-all-topics and the dynamic_bridge indeed created a subscriber without ros2 subscriber. 
**But still lost the previous messages.**

The sequence of running the nodes are following:
```
# terminal a
ros2 run ros1_bridge dynamic_bridge --brge-all-topics
```
```
# terminal b
ros2 run test sub_old_school.py
```
```
# terminal c
rosrun test pub.py
```
In terminal c:
[image]

In terminal b:
[image]

I lost the **message 0**. And if increase the frequency of pub will lost more messages.

I have changed qos for bridge.
[image]

and set the qos for ros2 subscriber:
```
import rclpy

from std_msgs.msg import String
from rclpy.qos import QoSProfile
from rclpy.qos import QoSReliabilityPolicy

g_node = None

def chatter_callback(msg):
    global g_node
    g_node.get_logger().info(
        'I heard: "%s"' % msg.data)

def main(args=None):
    global g_node
    rclpy.init(args=None)
    g_node = rclpy.create_node('minimal_subscriber')

    qos_profile = QoSProfile(
            depth = 100,
            reliability = QoSReliabilityPolicy.RMW_QOS_POLICY_RELIABILITY_RELIABLE
            )

    subscription = g_node.create_subscription(String, 'topic', chatter_callback,qos_profile)
    subscription  # prevent unused variable warning

    while rclpy.ok():
        rclpy.spin_once(g_node)
    g_node.get_logger().info('The spin was ended!')
    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    g_node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

I don't know why the messages from ros2 to ros1 will not be lost but from ros1 to ros2 will lost.

dirk-thomas, 2020-04-07:
Establishing connections between two endpoints is not instant and since that takes a non-zero amount of time messages published before that has happened will not be delivered. So what you are seeing is expected behavior in an asynchronous distributed system.

lx852357, 2020-04-07:
@dirk-thomas @fujitatomoya Thank you for your answers.

closed 2020-04-07 by lx852357