ros2/ros1_bridge #261: Error while building the bridge with Docker image

MaximeAubanel, 2020-05-11:
## Bug report

**Required Info:**

- Operating System:
  - OSX but building and running in Docker -->
- Installation type:
  - binaries
- Version or commit hash:
  - 5f70a38cf4ad9f11767350052552c260dd5301e9
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

- Build this docker.
```
FROM osrf/ros:eloquent-ros1-bridge
LABEL maintainer="Maxime Aubanel <maxime@rigi.tech>"

RUN apt-get update \
  && apt-get install -y ssh \
    build-essential \
    gcc \
    g++ \
    gdb \
    clang \
    cmake \
    rsync \
    nano \
    tar \
    python \
    ros-eloquent-launch* \
  && apt-get clean

ENTRYPOINT ["/ros_entrypoint.sh"]

WORKDIR "/home/workspace"

CMD /bin/bash
```

- Run this docker.

*inside the docker*
- Clone the repository && cd inside

- colcon build --symlink-install --packages-skip ros1_bridge

- source /opt/ros/melodic/setup.bash

- source /opt/ros/eloquent/setup.bash

- colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure

#### Expected behavior

Normal building

#### Actual behavior

Error while building such as : 
```
error: ‘rclcpp::FutureReturnCode’ has not been declared
error: ‘MessageInfo’ in namespace ‘rclcpp’ does not name a type
error: request for member ‘get_rmw_message_info’ in ‘msg_info’, which is of non-class type ‘const int’
```

#### Additional information

I had to add the installation of ros-eloquent-launch* to the official image otherwise I was getting another that cmake error:

```
CMake Error at CMakeLists.txt:258 (find_package):
  By not providing "Findlaunch_testing_ament_cmake.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "launch_testing_ament_cmake", but CMake did
  not find one.

  Could not find a package configuration file provided by
  "launch_testing_ament_cmake" with any of the following names:

    launch_testing_ament_cmakeConfig.cmake
    launch_testing_ament_cmake-config.cmake

  Add the installation prefix of "launch_testing_ament_cmake" to
  CMAKE_PREFIX_PATH or set "launch_testing_ament_cmake_DIR" to a directory
  containing one of the above files.  If "launch_testing_ament_cmake"
  provides a separate development package or SDK, be sure it has been
  installed.
```

dirk-thomas, 2020-05-11:
Please re-fill your ticket in the repository https://github.com/osrf/docker_images/ which contains the logic how this Docker image is created.

closed 2020-05-11 by dirk-thomas

gavanderhoorn, 2020-05-11:
Could it be that @MaximeAubanel is attempting to build `master` in an Eloquent container?

dirk-thomas, 2020-05-11:
> Could it be that @MaximeAubanel is attempting to build `master` in an Eloquent container?

If that is the case you need to use the `eloquent` branch of this repository.

MaximeAubanel, 2020-05-11:
Indeed, I had to checkout to the eloquent branch. It is not stated anywhere so ... might be nice to add it to the readme 😉