ros2/ros1_bridge #262: Example with simple_bridge doesn't work

MaximeAubanel, 2020-05-12:
## Bug report

**Required Info:**

- Operating System:
  - OSX but building and running in Docker
- Installation type:
  - binaries
- Version or commit hash:
  - 5f70a38cf4ad9f11767350052552c260dd5301e9
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

- Follow the tutorial in the README.md and use simple_bridge instead of dynamic_bridge

#### Expected behavior

Data correctly transmitted from ROS1 talker node to ROS2 listener node.

#### Actual behavior

Nothing is being received by the listener.

#### Additional information

Can do the PR if necessary. The fix is to change the the QoS policies and replace `rclcpp::SensorDataQoS()` by 10 as it's the one being used everywhere.

jacobperron, 2020-05-15:
@MaximeAubanel Sounds like you are building from source on top of a binary installation. Can you clarify which version of ROS 2 you have installed (e.g. Dashing or Eloquent)?

MaximeAubanel, 2020-05-16:
What do you mean by "building from source on top of a binary installation" ?

I'm using Eloquent.

jacobperron, 2020-05-16:
Sorry, I was just confused by:

> OSX but building and running in Docker

dirk-thomas, 2020-05-19:
@MaximeAubanel Please provide reproducible steps.

MaximeAubanel, 2020-05-20:
As I said in my first comment,  you just need to follow the tutorial.

Basically, you need to have to follow your README.md. When reaching the first example https://github.com/ros2/ros1_bridge#example-1a-ros-1-talker-and-ros-2-listener, instead of running `ros2 run ros1_bridge dynamic_bridge` run `ros2 run ros1_bridge simple_bridge`. Is that enough ?

dirk-thomas, 2020-05-20:
Since the fix was trivial I created #264 to resolve this.

PR #264 (2020-05-20): use reliable publisher in simple bridge
Fixes #262.

As already done in the [simple_bridge_1_to_2](https://github.com/ros2/ros1_bridge/blob/37a06eb64c97571766e245b6b2784a99c1c4df69/src/simple_bridge_1_to_2.cpp#L52). Otherwise subscribes must use best effort and can't use reliable.
closed 2020-05-20 by dirk-thomas