ros2/ros1_bridge #265: ROS2-side server bridge stays open after the server is destroyed

audrow, 2020-05-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal

- Installation type:
  - From source 

- Version or commit hash:
  - 5f70a38cf4ad9f11767350052552c260dd5301e9

- DDS implementation: 
  - Fast-RTPS

- Client library:
  - rclcpp and rclpy

#### Steps to reproduce issue

Launch a ROS1 master:
```
# Shell A:
. <ros-install-dir>/setup.bash
roscore -p 11311
```

Launch the `dynamic_bridge`:
```
# Shell B:
. <ros-install-dir>/setup.bash
. <ros2-install-dir>/setup.bash
export ROS_MASTER_URI=http://localhost:11311
ros2 run ros1_bridge dynamic_bridge
```
Launch a server on the ROS2-side:

```
# Shell C:
. <ros2-install-dir>/setup.bash
ros2 run demo_nodes_py add_two_ints_server
```

Launch a client on the ROS1-side:

```
# Shell D:
. <ros-install-dir>/setup.bash
export ROS_MASTER_URI=http://localhost:11311
rosrun rospy_tutorials add_two_ints_client
```

Now, stop the server in shell C and run a server on the ROS1-side:

```
# Shell D (continued):
rosrun rospy_tutorials add_two_ints_server
```

Launch a client on the ROS2 side:

```
# Shell C (continued):
ros2 run demo_nodes_py add_two_ints_client
```

The service won't be found by the client and you'll see the message `service not available, waiting again...`.

In my experience, the `dynamic_bridge` has to be restarted before I can start a service with the same service name.  

You can try going in the other direction: start with a service on ROS1 before trying a service on ROS2 and it should work.  Notice that on the bridge, it says `Created 2 to 1 bridge for service /add_two_ints` when the server is created and then `Removed 2 to 1 bridge for service /add_two_ints` when the server is stopped.  On the ROS2 side, the bridge is created, but the bridge is not removed when the service is stopped.  I have found this to be true for both the C++ and Python examples.

#### Expected behavior
That I'm able to create a server with the same name on either the ROS1-side or ROS2-side, close that server, and then create the server on the other side.

#### Actual behavior
I'm not able to discover the server created on the ROS1-side of the bridge after it has been created and the node has been deleted on the ROS2-side.  It seems that this is because the bridge for the ROS2 server is never removed.

dirk-thomas, 2020-05-22:
@audrow It would be great if you could test this case again with the two proposed patches and confirm that is solves the problem.

audrow, 2020-05-22:
@dirk-thomas Will do.

closed 2020-05-26 by dirk-thomas

audrow, 2020-05-26:
@dirk-thomas I can confirm that the two patches solve the problem.

PR #267 (2020-05-26): fix removing obsolete ROS 2 service bridges
Fixes #265. Requires ros2/rclcpp#1131.

CI builds testing `rclcpp` with only FastRTPS:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Packaging build with only FastRTPS: [image]