ros2/ros1_bridge #270: Failed to connect to user:35249

Michael-Equi, 2020-05-30:
- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 60f903c
- DDS implementation:
  - Fast-RTPS

Immediately after starting up the ros1_bridge dynamic_bridge node and after sourcing all the  appropriate setup scripts the node begins printing `Failed to connect to user:35249`. The port number references at the end seems to change between machines using the same environment. Everything else about the node seems to be working fine as data is still transmitted to and from the ROS2 and ROS1 nodes I am working with.

dirk-thomas, 2020-06-08:
The descriptions doesn't contain enough information to either investigate anything nor giving you any suggestions. Please include additional information, e.g. how to reproduce your issue.

Michael-Equi, 2020-06-08:
@dirk-thomas Steps to reproduce are building the version I mentioned in the comment from source and running the dynamic bridge. This occurs all on a single machine. I have been working with it on both ARM and X86 architectures on Ubuntu 18.04 and it seems to be an issue with the version. I am not sure if any newer versions, if they exist, have the same issue. It also does not happen ever time it runs but when it does I can find not correlation to anything going on with the rest of the system factor.

dirk-thomas, 2020-06-08:
This is still not sufficient to reproduce the problem. I don't ever see that message when running the `dynamic_bridge`: https://github.com/ros2/ros1_bridge/blob/60f903cb159ad4ef59b9b09badf71caa601cba04/src/dynamic_bridge.cpp#L399

For some reason the bridge is unable to connect to the node to retrieve information about its services. Maybe you are stopping nodes while the bridge is running? Anyway without extra context I can only speculate. Therefore you need to provide reproducible steps, step by step instructions what you run on the ROS 1 and ROS 2 side or even better a Dockerfile to just run the instructions.

ScottMcMichael, 2020-08-15:
I reproduced this problem with just a rostopic pub/ros2 topic echo pair and was able to fix it by setting these in each of the console instances:
export ROS_HOSTNAME=localhost
export IGN_IP=127.0.0.1
export ROS_MASTER_URI=http://localhost:11311

dirk-thomas, 2020-08-17:
Assuming this is just an environment setup problem I will go ahead and close this ticket for now. Please feel free to comment with more information if this is still a reproducible issue and the ticket can be reopened.

closed 2020-08-17 by dirk-thomas

berndpfrommer, 2021-03-05:
I had the same issue and it went away by simply restarting the roscore.