ros2/ros1_bridge #273: Failed to detect service while running in Docker container

atb033, 2020-06-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - eloquent
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

-  Run the ros1_bridge dynamic_bridge --bridge-all-topics in a Docker container
-  Run the roscpp add_two_ints server in a Docker container
-  Run the rclcpp add_two_ints client in a Docker container
-  Run the rclcpp add_two_ints client a few more times and notice that the service becomes unavailable

The necessary docker files and instructions for replicating this bug can be found [here](https://github.com/umdlife/ros1_bridge_bug_report).

#### Expected behavior

The add_two_ints client docker container must be able to detect the add_two_ints server even when it is restarted 

#### Actual behavior

The add_two_ints client was able to detect the server only during the first attempt

fujitatomoya, 2020-06-12:
@atb033 

service is constructed on topics, might be related to https://github.com/ros2/rmw_fastrtps/issues/349. The reason is when you initiate the container the same process id will be allocated, that leads to the same global id in dds.

after all, could you try to use Foxy? if the problem is related to https://github.com/ros2/rmw_fastrtps/issues/349, you should not meet the problem.

atb033, 2020-06-12:
@fujitatomoya 

Thanks for the quick response. I tried it with Foxy, and didn't face this problem

dirk-thomas, 2020-06-12:
I am going to close this ticket since the problem doesn't seem to be related to `ros1_bridge`.

closed 2020-06-12 by dirk-thomas