ros2/ros1_bridge #276: Failed to build from source: theora_image_transport unable to find opencv_calib3d dependency

cardboardcode, 2020-07-14:
## Bug report

**Required Info:**

- **Operating System**:
  - Ubuntu **18.04**, ROS2 **Eloquent**, ROS1 **Melodic**
- **Installation type**:
  - Build From Source. Using [the instructions provided](https://github.com/ros2/ros1_bridge/tree/eloquent#building-the-bridge-from-source) in `README.md`
- **Version or commit hash**:
  -  Version/branch: eloquent, commit hash 9cc654b
- **DDS implementation**:
  - Default Fast-RTPS 
- **Client library** (if applicable):
  - N/A

#### Steps to reproduce issue
```
mkdir -p ros1_bridge_ws/src && cd ros1_bridge_ws/src
git clone https://github.com/ros2/ros1_bridge.git --branch eloquent  --single-branch
cd ~/ros1_bridge_ws
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
```

#### Expected behavior
Colcon build passes.

#### Actual behavior
Colcon build fails with the following error:

```
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
CMake Error at /opt/ros/eloquent/share/theora_image_transport/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  By not providing "Findopencv_calib3d.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "opencv_calib3d", but CMake did not find one.

  Could not find a package configuration file provided by "opencv_calib3d"
  with any of the following names:

    opencv_calib3dConfig.cmake
    opencv_calib3d-config.cmake

  Add the installation prefix of "opencv_calib3d" to CMAKE_PREFIX_PATH or set
  "opencv_calib3d_DIR" to a directory containing one of the above files.  If
  "opencv_calib3d" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /opt/ros/eloquent/share/theora_image_transport/cmake/theora_image_transportConfig.cmake:38 (include)
  CMakeLists.txt:94 (find_package)

---
Failed   <<< ros1_bridge [5.18s, exited with code 1]

Summary: 0 packages finished [5.29s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge

```

#### Additional information

My goal is to build **ros1_bridge** from source so ROS2-ROS1 topic mapping can be done for custom interfaces/messages.

However, I am unable to build **ros1_bridge** from source.

I have verified the issue does not lie within **theora_image_transport**. When building **theora_image_transport** from source, I did not encounter this build failure.

Follow the steps below to reproduce how I built **theora_image_transport** from source.
```
mkdir -p test_ws/src && cd test_ws/src
git clone https://github.com/ros-perception/image_transport_plugins.git -- branch ros2 --single-branch

<Replace all instances of the following lines in the package directory>
Change #include "image_transport/simple_subscriber_plugin.hpp" to #include "image_transport/simple_subscriber_plugin.h"
Change #include "image_transport/simple_publisher_plugin.hpp" to #include "image_transport/simple_publisher_plugin.h"

[This is because, before ROS2 Foxy, .hpp has not been used yet. I am currently using ROS2 Eloquent.]

cd ~/test_ws
colcon build --symlink-install

```
N/A

#### Feature description
N/A
#### Implementation considerations
N/A

dirk-thomas, 2020-07-14:
The issue is in the `theora_image_transport` package. While it might build successfully it doesn't export its necessary dependencies. That is why the error happens within `/opt/ros/eloquent/share/theora_image_transport/cmake/theora_image_transportConfig.cmake`. Please file a ticket in the `theora_image_transport` repository instead.

closed 2020-07-14 by dirk-thomas

cardboardcode, 2020-07-14:
Thanks for the clarification. Will file the ticket soon.

cardboardcode, 2020-07-15:
Just to record on further progress on debugging this issue. 

Including it here since this still pertains to the build process of **ros1_bridge**.

## Tried removing **theora_image_transport** as a dependency during colcon build compilation

Attempt failed.

Manually removed **theora_image_transport** dependency by **inserting** the following after [**line 92** of `CMakeLists.txt`](https://github.com/ros2/ros1_bridge/blob/51bef560c6bef396872e110a0182440afd4ca4bd/CMakeLists.txt#L92).

```cmake
set(ros2_interface_packages "action_msgs"
"action_tutorials_interfaces"
"actionlib_msgs"
"builtin_interfaces"
"composition_interfaces"
"diagnostic_msgs"
"example_interfaces"
"gazebo_msgs"
"geometry_msgs"
"lifecycle_msgs"
"logging_demo"
"map_msgs"
"nav_msgs"
"pcl_msgs"
"pendulum_msgs"
"rcl_interfaces"
"rosgraph_msgs"
"rqt_py_common"
"sensor_msgs"
"shape_msgs"
"std_msgs"
"std_srvs"
"stereo_msgs"
"tf2_msgs"
"trajectory_msgs"
"unique_identifier_msgs"
"visualization_msgs")
```

Encountered Error
```
CMake Warning at CMakeLists.txt:208 (add_library):
  Cannot generate a safe runtime search path for target ros1_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/eloquent/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/eloquent/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/eloquent/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/eloquent/lib
    runtime library [libconsole_bridge.so.0.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/eloquent/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:179 (add_executable):
  Cannot generate a safe runtime search path for target simple_bridge_2_to_1
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libconsole_bridge.so.0.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/eloquent/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
[... 107 lines cut ...]
Call Stack (most recent call first):
  CMakeLists.txt:252 (custom_executable)

libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_theora_image_transport(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
libros1_bridge.so: undefined reference to `ros1_bridge::get_service_factory_theora_image_transport(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [dynamic_bridge] Error 1
make[1]: *** [CMakeFiles/dynamic_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_theora_image_transport(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
libros1_bridge.so: undefined reference to `ros1_bridge::get_service_factory_theora_image_transport(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [parameter_bridge] Error 1
make[1]: *** [CMakeFiles/parameter_bridge.dir/all] Error 2
libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_theora_image_transport(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
libros1_bridge.so: undefined reference to `ros1_bridge::get_service_factory_theora_image_transport(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [static_bridge] Error 1
make[1]: *** [CMakeFiles/static_bridge.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_bridge [28.4s, exited with code 2]
```

Backing away since removing the dependency only made the issue harder to disentangle.

cardboardcode, 2020-07-16:
Further progress on debugging this issue:

## Not a proper fix. But build now passes.

The superficial issue of getting it to compile can be resolved by:
1. Build **ros1_bridge** and **image_transport_plugins** from source.
```
[Open terminal]

mkdir -p ros1_bridge_ws/src && cd ros1_bridge_ws/src
git clone https://github.com/ros2/ros1_bridge.git --branch eloquent  --single-branch
git clone https://github.com/ros-perception/image_transport_plugins.git -- branch ros2 --single-branch
cd ~/ros1_bridge_ws
colcon build --symlink-install --packages-skip ros1_bridge

[Close terminal]
[Open terminal]

cd ~/ros1_bridge_ws
source /opt/ros/melodic/setup.bash
source /opt/ros/eloquent/setup.bash
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
```
2. Replace the following [line 96 to 102 of `image_transport_plugins\theora_image_transport\CMakeLists.txt`](https://github.com/ros-perception/image_transport_plugins/blob/0565f23dc955fa2d56cffcfebc63de5a97cd005f/theora_image_transport/CMakeLists.txt#L96-L102)

```cmake
ament_export_dependencies(rosidl_default_runtime
  ${OpenCV_LIBRARIES}
  ${PC_OGG_LIBRARIES}
  ${PC_THEORA_LIBRARIES}
  ${PC_THEORAENC_LIBRARIES}
  ${PC_THEORADEC_LIBRARIES}
  ${LIBRARY_NAME})
```
Replace the above lines with:
```cmake
ament_export_dependencies(rosidl_default_runtime
  OpenCV
  PkgConfig
)
```
Will continue to update here once a proper fix is found.

cardboardcode, 2020-07-16:
Further progress on debugging this issue.

# Fix No.2
## Still not a proper fix but it is better now.

The original goal is to be able to compile **ros1_bridge** from source in order to conduct topic mapping for custom ROS messages.

With this fix, I am able to do the following:
1. Compile ros1_bridge from source while avoiding the build error mentioned in this issue thread.
2. Be able to publish custom ROS message in ROS2 and receive in ROS1.

Follow the steps below to achieve the aforementioned 2 objectives:

### Download ros1_bridge & image_transport_plugins ROS2 package
```
mkdir -p ros1_bridge_ws/src && cd ros1_bridge_ws/src
git clone https://github.com/ros2/ros1_bridge.git --branch eloquent  --single-branch
git clone https://github.com/ros-perception/image_transport_plugins.git -- branch ros2 --single-branch
```
### Modify image_transport_plugins
 ```
cd ros1_bridge_ws/src/image_transport_plugins
<Replace all instances of the following lines in the package directory>
Change #include "image_transport/simple_subscriber_plugin.hpp" to #include "image_transport/simple_subscriber_plugin.h"
Change #include "image_transport/simple_publisher_plugin.hpp" to #include "image_transport/simple_publisher_plugin.h"

[This is because, before ROS2 Foxy, .hpp has not been used yet. I am currently using ROS2 Eloquent.]

<Replace the following lines 96 to 102 of image_transport_plugins\theora_image_transport\CMakeLists.txt with the ones below>
ament_export_dependencies(rosidl_default_runtime
  OpenCV
  PkgConfig
)
```

### Include your custom message as a ROS2 package.
```
cd ros1_bridge_ws/src
<Create your custom message package. Ensure that the package name has _msgs or _interface.>

cd YOUR_ROS1_WORSPACE/src
<Create your custom message package. Ensure it has the same name as the one in ROS2.>
<Build and source your ROS1 workspace>

```
Note that this is how **ros1_bridge** match ROS2-ROS1 message pairs together. See [this link for the evidence](https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst#how-are-ros-1-and-2-messages-associated-with-each-other)

### Build everything in your ROS2 workspace, except ros1_bridge, first
```
cd ~/ros1_bridge_ws
source /opt/ros/eloquent/setup.bash
colcon build --symlink-install --packages-skip ros1_bridge
``` 
### Build ros1_bridge
```
source /opt/ros/melodic/setup.bash
source /opt/ros/eloquent/setup.bash
source ~/YOUR_ROS1_WORSPACE/devel/setup.bash # The ROS1 workspace that contains the custom message.
source ~/ros1_bridge_ws/devel/setup.bash # The ROS2 workspace that contains the custom message.
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
```

That's it. That was how I "fixed" my issue, at least temporarily, and achieved my original goal.

## How do you know it worked?
Don't close that terminal you used for building your ros1_bridge yet. But if you did, please do the following:
```
source /opt/ros/melodic/setup.bash
source /opt/ros/eloquent/setup.bash
source ~/YOUR_ROS1_WORSPACE/devel/setup.bash # The ROS1 workspace that contains the custom message.
source ~/ros1_bridge_ws/devel/setup.bash # The ROS2 workspace that contains the custom message.
```
If you had not close that terminal, 
```
ros2 run ros1_bridge dynamic_bridge --print-pairs
```
It has worked properly if you see your custom message in the list of matching pairs printed on the terminal.

## To run ros1_bridge with support for custom message
```
[Open new terminal]
ros2 run ros1_bridge dynamic_bridge --bridge-all-topics

[Open new terminal]
source /opt/ros/eloquent/setup.bash
ros2 run [THE ROS2 NODE THAT IS PUBLISHING THE CUSTOM MESSAGE]

[Open new terminal]
source /opt/ros/melodic/setup.bash
roscore

[Open new terminal]
source /opt/ros/melodic/setup.bash
rostopic list
[The topic with custom message in ROS2 should now appear.]
```