ros2/ros1_bridge #277: Dynamic-bridge throw error and exit when returning false

MaximeAubanel, 2020-07-19:
## Bug report

**Required Info:**

- Operating System:
  - Docker on Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Installed the bridge via `ros-foxy-ros1-bridge`
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - Roscpp & Rclcpp

When calling a service using ROS2 as client to ROS1 as server, using the dynamic-bridge. The bridge exit after throwing an error:

```
terminate called after throwing an instance of 'std::runtime_error'
what():  Failed to get response from ROS 1 service /command_X
```

I figured out something. This happens only when the service return false, like that:

```
bool aarming_service(std_srvs::Empty::Request& request, std_srvs::Empty::Response& response) {
  return false;
}
```

If it returns true, the error does not appear and the bridge works properly.

Why is that ?

dirk-thomas, 2020-07-20:
Unfortunately ROS 2 servers don't have a way to communicate this failure generically to the client. Until ros2/ros2#921 is implemented there is no much the ros1_bridge can do about it (except just ignoring the service call and letting the client wait forever for a response which isn't really an option either).

MaximeAubanel, 2020-07-20:
Thank you ! Was not aware of that :)

closed 2020-07-20 by MaximeAubanel