ros2/ros1_bridge #278: Tried to advertise on topic [/rosout] with latch=0 but the topic is already advertised with latch=1

Tahir-Rasheed-437, 2020-07-20:
Hey guys. i am currently using vrep simulation. i have recently started using ROS2 foxy.  Earlier, I have used ROS2 Dashing and eloquent with melodic and used ROS1bridge for successful communication. No issues there. Now i have started using foxy with noetic. Once i run ros2 run ros1_bridge dynamic_bridge --bridge-all-topics, i get the following error with a warning. 

[ERROR] [1595237693.068410353]: Tried to advertise on topic [/rosout] with latch=0 but the topic is already advertised with latch=1
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
[INFO] [1595237693.071945376] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Log to ROS 2 rcl_interfaces/msg/Log (showing msg only once per type)
[WARN] [1595237693.072375865] [ros_bridge]: Message from ROS 2 rcl_interfaces/msg/Log failed to be passed to ROS 1 rosgraph_msgs/Log because the ROS 1 publisher is invalid (showing msg only once per type)

The communication starts and i can see the topics on both sides. Problem comes when i try to use the tf's in ROS1 generated from the ROS2 side.  There is a mismatch of time_stamp. Rviz give me the following error:
TF_REPEATED_DATA ignoring data with redundant timestamp for frame

I have to use some packages on ROS1 side that uses the tf's generated on ROS2 side. Any idea what might be the issue here?

Secondly, i have some static tf's generated from ROS2 side (foxy) that also cannot be seen on ROS1. Previously for ROS2 Dashing, it could see the static publishers in ROS1 but its not the case anymore in foxy. Any idea what additionally needed to be in the bridge to get the static publishers? or it will be seen once the above error is solved
Thanks

dirk-thomas, 2020-07-20:
This seems to be related to ros/ros_comm#1991 and will likely be fixed by the proposed patch in `ros_comm` in Noetic.

dirk-thomas, 2020-08-06:
Since the fix will on the ROS 1 side has been released into Noetic I will go ahead and close this ticket.

closed 2020-08-06 by dirk-thomas