ros2/ros1_bridge #280: Gazebo /clock messages not recieved in ROS2 correctly when sent through ros1_bridge

michalbahnik, 2020-07-30:
Hi, I experience some strange behavior connected to using Gazebo and ros1_bridge together. If I try to send /clock through the bridge to ROS2, big part of the messages are not recieved. I am not sure, if error is in the bridge or in gazebo plugin, but ROS1 topic is published (and can be recieved) correctly, unlike messsages transferred via bridge, so I suspect, something bad happens in bridge.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.9.3-1focal.20200722.060820
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclpy
- ROS versions
  - Noetic
  - Foxy
- Gazebo version
  - Gazebo 11

#### Steps to reproduce issue
1) Run test Gazebo world with ROS
```
rosrun gazebo_ros gazebo
```
2) Run ros1_bridge
```
ros2 run ros1_bridge dynamic_bridge /clock
```
3) Run minimal subscriber for /clock topic.
```
#!/usr/bin/env python
import rclpy
from rclpy.node import Node

from rosgraph_msgs.msg import Clock

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            Clock,
            '/clock',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg: Clock):
        self.get_logger().info(f'I heard: "{msg}"')

def main(args=None):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()

    rclpy.spin(minimal_subscriber)

    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```
4) Check frequency with
```
rostopic hz /clock
```
and
```
ros2 topic hz /clock
```

#### Expected behavior
Recieving frequency (ROS2) expected to be similar to publishing (ROS1) frequency ~ 1000 Hz.

#### Actual behavior
Messages recieved (ROS2) with much lower frequency (avg < 200).

#### Additional information
Clock messages are recieved correctly for some time (< 1 s) and then lag of few houndereds of miliseconds happens.

Note that if I run simple publisher (according to oficial tutorials), frequency matches. If I try this while Gazebo simulation is running (and publishing), frequency decreases again.

Thanks for any advises and/or fixes!

dirk-thomas, 2020-08-14:
I tried to reproduce your problem with the following shells:

* ROS 1: `roscore`
* ROS 1: `rosrun gazebo_ros gazebo`
* ROS 1 and 2: `ros2 run ros1_bridge dynamic_bridge` (note: this executable does not accept a topic name as an argument, your `/clock` argument was simply ignored)
* ROS 2: `ros2 topic echo /clock rosgraph_msgs/msg/Clock`
* ROS 2: `ros2 topic hz /clock`

And I get an average rate of > 980 Hz.

Please try the above steps to check if the problem is related to your custom subscriber or something else.

michalbahnik, 2020-08-14:
Hello @dirk-thomas , thank you for your answer! I replicated steps you posted above and I reached average rate of about 970 Hz. Then I also tried to use the listener script from my original answer and the performance was very similar. I believe there was something wrong with my ROS Noetic installation (including all gazebo packages), because I also experienced some other issues and in the end (after dealing with this issue) I reinstalled it. After reinstallation, ros1_bridge and gazebo works as expected.

closed 2020-08-14 by michalbahnik