ros2/ros1_bridge #283: support for fixed size array in service calls

cyrilleberger, 2020-08-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04/20.04
- Installation type:
  - binaries for everything ros1_bridge
- Version or commit hash:
  - dasing and foxing
- DDS implementation:
  - default one

#### Steps to reproduce issue
Have a service call with definition with a fixed size array:

```
---
float64[6] transform
```

#### Expected behavior

Match between ROS1 and ROS2.

#### Actual behavior

Fails to match between ROS1 and ROS2.

#### Additional information

I submited a working patch https://github.com/ros2/ros1_bridge/pull/242 but I am lacking the time to finish adjusting the styles and conflicts that keep coming up.

klaxalk, 2021-01-12:
I second this issue.

klaxalk, 2021-12-12:
Any update on this?

gbiggs, 2021-12-12:
Unfortunately I don't think anyone is working on this currently. If you have time to work on it, I recommend starting from the existing patch and completing that work.