ros2/ros1_bridge #284: Allow exclusion of / ignoring of msg types

norro, 2020-09-07:
Allow to exclude certain message types and/or packages from the rosbridge build. This could be done at cmake time, e.g., through cmake configuration options.

AFAIR, this can be helpful to:

- ignore message types that are not compatible between ROS 1 und ROS 2 or don't have a proper, therefore breaking the rosbridge build
- reduce rosbridge build time

norro, 2020-09-09:
I think both a blacklist as well as a whitelist of types might be reasonable.
What do you think?

clalancette, 2020-09-24:
Hi @norro ,

This sounds like a good enhancement to the bridge.  Could you create a pull request implementing this feature? Thanks.

norro, 2020-09-28:
Sure, I will give it a shot.

norro, 2020-11-17:
I think a *package* blacklist/whitelist is more suited for the use-case of reducing build time, whereas a *message* blacklist/whitelist is more suited for the use-case of ignoring message types that are not compatible between ROS 1 und ROS 2.