ros2/ros1_bridge #290: Error buiding ros1 bridge

Schloern93, 2020-11-09:
Hello together,

System Ubuntu 18.04
Ros Environment
1. Ros dashing 
2. Ros melodic

The following are the steps I have tried

1. Get ros2_bridge with git clone into my ros2 dashing workspace
2. source ros2 dashing environment
3. build workspace without the ros1_bridge 
                     colcon build --symlink-install --packages-skip ros1_bridge
4. source ros1 melodic environment in the same terminal
5. source ros2 dashing workspace 
6 build the ros1_bridge 
                      colcon build --symlink-install --packages-select ros1_bridge --cmake-               force-configure

Now I got the following building error:
----------------------------------------------------------------------------------
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
--- stderr: ros1_bridge                              
CMake Warning at CMakeLists.txt:174 (add_library):
  Cannot generate a safe runtime search path for target ros1_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:145 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:191 (custom_executable)

CMake Warning at CMakeLists.txt:145 (add_executable):
  Cannot generate a safe runtime search path for target parameter_bridge
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
[... 237 lines cut ...]
/home/florian/ros2_platform/ros2_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:142:64: error: ‘MessageInfo’ in namespace ‘rclcpp’ does not name a type
       void(const typename ROS2_T::SharedPtr msg, const rclcpp::MessageInfo & msg_info)> callback;
                                                                ^~~~~~~~~~~
/home/florian/ros2_platform/ros2_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::ros2_callback(typename ROS2_T::SharedPtr, const int&, ros::Publisher, const string&, const string&, rclcpp::Logger, rclcpp::PublisherBase::SharedPtr)’:
/home/florian/ros2_platform/ros2_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:223:19: error: request for member ‘get_rmw_message_info’ in ‘msg_info’, which is of non-class type ‘const int’
         &msg_info.get_rmw_message_info().publisher_gid,
                   ^~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/ros1_bridge.dir/generated/robot_interface__srv__RobotControl__factories.cpp.o] Error 1
make[1]: *** [CMakeFiles/ros1_bridge.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_bridge [19.4s, exited with code 2]

Summary: 0 packages finished [19.5s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge

---------------------------------------------------------------------------------------
The Error:
/home/florian/ros2_platform/ros2_ws/src/ros1_bridge/test/test_ros2_client.cpp:37:13: error: ‘rclcpp::FutureReturnCode’ has not been declared
     rclcpp::FutureReturnCode::SUCCESS)

Is it possible that the code is too new for ros dashing?

Thanks in advance

clalancette, 2020-11-09:
Yes, if you are building against Dashing, you have to use the `dashing` branch of `ros1_bridge`.  Can you please try that and report back?

Schloern93, 2020-11-10:
Thanks for the quick response.
I can build the package now. But I still get an error. File was attached.
But I can run the examples without errors.

[error_ building_ros1_bridge.txt](https://github.com/ros2/ros1_bridge/files/5517438/error_.building_ros1_bridge.txt)

clalancette, 2020-11-10:
Interesting, not sure why you are getting those warnings.  We are overlaying urdfdom libraries on top of the system ones, so that may be why CMake is complaining.

But in any case, those don't look like fatal warnings, and as you say your examples are working.  So I'm going to go ahead and close this out for now.

closed 2020-11-10 by clalancette