ros2/ros1_bridge #291: Ros bridge does not translate service messages

Schloern93, 2020-11-11:
Hallo,

i have created a Ros1 service and a Ros2 service according to the following instructions
Ros1:
http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28c%2B%2B%29
Ros2:
https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Service-And-Client/

I have now built the bridge in a separate package as follows.
1. source Ros2 environment
2. source Ros1 environment
4 Source Service package ros1
5. source service package ros2
6. building the Ros bridge in the separate package
Then I started the Ros bridge as described in the manual. For the examples of the subscriber and the publisher the bridge works.
But if I start the service over Ros1 and the client over Ros 2 I can't receive any messages. 
I have attached screenshots of the terminals for starting the bridge and the service
Did I do something wrong when I built the bridge?
Many thanks in advance

Translated with www.DeepL.com/Translator (free version)
[image]
[image]
[image]
[image]

iuhilnehc-ynos, 2020-11-17:
@Schloern93 

Please make sure checking the `pairs` information will show the following message

```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs | grep begin
  - 'example_interfaces/srv/AddTwoInts' (ROS 2) <=> 'beginner_tutorials/AddTwoInts' (ROS 1)
```

> Did I do something wrong when I built the bridge?

Yes, I think that you could rebuild [example_interfaces](https://github.com/ros2/example_interfaces) after appending a mapping rule for your test into [mapping_rules.yaml](https://github.com/ros2/example_interfaces/blob/34f11b205aa2f5b79abe17d9c561ce53291128b3/mapping_rules.yaml)

```
-
  ros1_package_name: 'beginner_tutorials'
  ros1_service_name: 'AddTwoInts'
  ros2_package_name: 'example_interfaces'
  ros2_service_name: 'AddTwoInts'
```

Schloern93, 2020-11-17:
I solved the problem. 
The problem was that I had built the ros1 workspace on kinetic. After I rebuilt it with melodic and built the ros_bridge it worked. 
The bridge now translates the messages to the ros kinetic workspace on my resberry pi.

closed 2020-11-17 by Schloern93

supergirl-os, 2022-03-08:
I built the ros_bridge on noetic, but it appears the same problem like you. I wonder if this is because the wrong vesion of ros. Or could you please tell me exactly how you solved this problem?