ros2/ros1_bridge #313: Build error: non-constant condition for static assertion

m2-farzan, 2021-05-05:
## Bug report

**Required Info:**

- Operating System:
  - Archlinux (May 6, 2021) **[GCC 10.2.0]**
- Installation type:
  - from source
- Version or commit hash:
  - foxy branch (3209bc77c77b072013b05bcc6690a24a6bd16aeb)  —But I have the same problem with master.
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
source /opt/ros/noetic/setup.bash
colcon build --merge-install --packages-up-to ros1_bridge --cmake-force-configure
```

#### Expected behavior
Build successfully

#### Actual behavior
These errors:
```
/ros2-pkg/ros2-foxy-PKGBUILD/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_1_to_2(const ROS1_T&, ROS2_T&) [with ROS1_T = geometry_msgs::AccelWithCovariance_<std::allocator<void> >; ROS2_T = geometry_msgs::msg::AccelWithCovariance_<std::allocator<void> >]’:
/ros2-pkg/ros2-foxy-PKGBUILD/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:69:34: error: non-constant condition for static assertion
   69 |     (ros2_msg.covariance.size()) >= (ros1_msg.covariance.size()),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/ros2-pkg/ros2-foxy-PKGBUILD/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:69:34: error: ‘ros2_msg’ is not a constant expression
/ros2-pkg/ros2-foxy-PKGBUILD/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_2_to_1(const ROS2_T&, ROS1_T&) [with ROS1_T = geometry_msgs::AccelWithCovariance_<std::allocator<void> >; ROS2_T = geometry_msgs::msg::AccelWithCovariance_<std::allocator<void> >]’:
/ros2-pkg/ros2-foxy-PKGBUILD/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:100:34: error: non-constant condition for static assertion
  100 |     (ros1_msg.covariance.size()) >= (ros2_msg.covariance.size()),
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/ros2-pkg/ros2-foxy-PKGBUILD/src/build/ros1_bridge/generated/geometry_msgs__msg__AccelWithCovariance__factories.cpp:100:34: error: ‘ros2_msg’ is not a constant expression

```

### Additional information
The problem went away when I reverted the commit 81b7610568286ec7b390c64cf6207b362d0a6550. I only had to run this:
```
git revert 81b7610568286ec7b390c64cf6207b362d0a6550 --no-edit
```

Thanks!

iuhilnehc-ynos, 2021-06-03:
```
std::array<int, 4> a;
sizeof(a)   // is bad
a.size()     // not working for `static_assert` while using g++10
std::tuple_size<decltype(a)>::value         // might be the correct way in the future.
```

xlla, 2022-04-24:
same issue here in galactic branch.

<pre>
/Users/xlla/ros2_foxy/build/ros1_bridge/generated/shape_msgs__msg__Plane__factories.cpp:61:6: error: static_assert expression is not an integral constant expression
    (ros2_msg.coef.size()) >= (ros1_msg.coef.size()),
    ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/xlla/ros2_foxy/build/ros1_bridge/generated/shape_msgs__msg__Plane__factories.cpp:84:11: error: implicit instantiation of undefined template 'std::__1::tuple_size<boost::array<double, 4> >'
    (std::tuple_size<decltype(ros1_msg.coef)>::value) >= (std::tuple_size<decltype(ros2_msg.coef)>::value),
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tuple:25:49: note: template is declared here
template <class _Tp> class _LIBCPP_TEMPLATE_VIS tuple_size;
</pre>

using method mentioned by @iuhilnehc-ynos give another error:
<pre>
/Users/xlla/ros2_foxy/build/ros1_bridge/generated/shape_msgs__msg__Plane__factories.cpp:84:11: error: implicit instantiation of undefined template 'std::__1::tuple_size<boost::array<double, 4> >'
    (std::tuple_size<decltype(ros1_msg.coef)>::value) >= (std::tuple_size<decltype(ros2_msg.coef)>::value),
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tuple:25:49: note: template is declared here
template <class _Tp> class _LIBCPP_TEMPLATE_VIS tuple_size;
                                                ^

</pre>

xlla, 2022-04-24:
change static_assert to assert pass the build, I no know the final effect.

<pre>
Starting >>> ros1_bridge
[Processing: ros1_bridge]                             
[Processing: ros1_bridge]                                     
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]      
Finished <<< ros1_bridge [5min 57s]
                           
Summary: 1 package finished [5min 58s]
  1 package had stderr output: ros1_bridge
</pre>