ros2/ros1_bridge #317: Failed to colcon build the ros1_bridge

BananaGorilla98, 2021-06-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04, ROS2 Galactic, ROS Noetic
- Installation type:
  - Binaries
- Version or commit hash:
  - 
- DDS implementation:
  - Default FAST-RTPS
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
source /opt/ros/noetic/setup.bash
source /opt/ros/galactic/local_setup.bash
source ./install/local_setup.bash
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
```

#### Expected behavior
The build success. 
#### Actual behavior
The colcon build failed. The error code shows 
```
/usr/include/c++/9/variant:1268:8: error: no type named ‘type’ in ‘struct std::enable_if<false, void>’
 1268 |  using __accepted_type = __to_type<__accepted_index<_Tp>>;
```
and 
```
/opt/ros/galactic/include/rclcpp/node_impl.hpp:100:47:   required from ‘std::shared_ptr<SubscriptionT> rclcpp::Node::create_subscription(const string&, const rclcpp::QoS&, CallbackT&&, const rclcpp::SubscriptionOptionsWithAllocator<AllocatorT>&, typename MessageMemoryStrategyT::SharedPtr) [with MessageT = mavros_msgs::msg::DebugValue_<std::allocator<void> >; CallbackT = std::function<void(std::shared_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> > >, const rmw_message_info_t&)>&; AllocatorT = std::allocator<void>; CallbackMessageT = mavros_msgs::msg::DebugValue_<std::allocator<void> >; SubscriptionT = rclcpp::Subscription<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::allocator<void>, rclcpp::message_memory_strategy::MessageMemoryStrategy<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::allocator<void> > >; MessageMemoryStrategyT = rclcpp::message_memory_strategy::MessageMemoryStrategy<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::allocator<void> >; std::string = std::__cxx11::basic_string<char>; typename MessageMemoryStrategyT::SharedPtr = std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::allocator<void> > >]’
/home/engyeankhai/garuda_copilot_ros_wkspace/garuda-copilot-ros-jetson-branch/garuda-copilot-ros/copilot-suite/copilot-bridge/src/ros1_bridge/include/ros1_bridge/factory.hpp:129:48:   required from ‘rclcpp::SubscriptionBase::SharedPtr ros1_bridge::Factory<ROS1_T, ROS2_T>::create_ros2_subscriber(rclcpp::Node::SharedPtr, const string&, const rmw_qos_profile_t&, ros::Publisher, rclcpp::PublisherBase::SharedPtr) [with ROS1_T = mavros_msgs::DebugValue_<std::allocator<void> >; ROS2_T = mavros_msgs::msg::DebugValue_<std::allocator<void> >; rclcpp::SubscriptionBase::SharedPtr = std::shared_ptr<rclcpp::SubscriptionBase>; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>; std::string = std::__cxx11::basic_string<char>; rmw_qos_profile_t = rmw_qos_profile_t; rclcpp::PublisherBase::SharedPtr = std::shared_ptr<rclcpp::PublisherBase>]’
/home/engyeankhai/garuda_copilot_ros_wkspace/garuda-copilot-ros-jetson-branch/garuda-copilot-ros/copilot-suite/copilot-bridge/src/ros1_bridge/include/ros1_bridge/factory.hpp:112:3:   required from here
/opt/ros/galactic/include/rclcpp/any_subscription_callback.hpp:185:25: error: no match for ‘operator=’ (operand types are ‘rclcpp::detail::AnySubscriptionCallbackHelper<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::allocator<void> >::variant_type’ {aka ‘std::variant<std::function<void(const mavros_msgs::msg::DebugValue_<std::allocator<void> >&)>, std::function<void(const mavros_msgs::msg::DebugValue_<std::allocator<void> >&, const rclcpp::MessageInfo&)>, std::function<void(std::unique_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::default_delete<mavros_msgs::msg::DebugValue_<std::allocator<void> > > >)>, std::function<void(std::unique_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::default_delete<mavros_msgs::msg::DebugValue_<std::allocator<void> > > >, const rclcpp::MessageInfo&)>, std::function<void(std::shared_ptr<const mavros_msgs::msg::DebugValue_<std::allocator<void> > >)>, std::function<void(std::shared_ptr<const mavros_msgs::msg::DebugValue_<std::allocator<void> > >, const rclcpp::MessageInfo&)>, std::function<void(const std::shared_ptr<const mavros_msgs::msg::DebugValue_<std::allocator<void> > >&)>, std::function<void(const std::shared_ptr<const mavros_msgs::msg::DebugValue_<std::allocator<void> > >&, const rclcpp::MessageInfo&)>, std::function<void(std::shared_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> > >)>, std::function<void(std::shared_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> > >, const rclcpp::MessageInfo&)> >’} and ‘rclcpp::detail::SubscriptionCallbackTypeHelper<mavros_msgs::msg::DebugValue_<std::allocator<void> >, std::function<void(std::shared_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> > >, const rmw_message_info_t&)>, void>::callback_type’ {aka ‘std::function<void(std::shared_ptr<mavros_msgs::msg::DebugValue_<std::allocator<void> > >, const rmw_message_info_t&)>’})
  185 |       callback_variant_ = static_cast<typename scbth::callback_type>(callback);
      |       ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
#### Additional information
May I know how to solve this issue?

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2021-07-01:
I'm not able to reproduce this with the standard bridge; it all compiles fine for me.  Can you give more information about your custom workspace and what is failing to compile there?

BananaGorilla98, 2021-07-14:
Hi, sorry for the late reply. We managed to solve the problem and built it successfully by cloning a fresh ros1_bridge. But I still can't explain why it works and why the same ros1_bridge package in our workspace fail in the first place.

clalancette, 2022-02-03:
OK, given the last response I'm going to close this out.

closed 2022-02-03 by clalancette