ros2/ros1_bridge #318: Colcon build failed

larsvlent, 2021-06-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Source
- Version or commit hash:
  - ROS2: dashing. ROS1: Melodic
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
source /opt/ros/dashing/setup.bash
source /opt/ros/melodic/setup.bash
. install/setup.bash
```

```

#### Expected behavior
A succelfull build

#### Actual behavior
Colcon build failed:
I have this:
`    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib`

and:
`CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target dynamic_bridge
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/dashing/lib]
      dir 1 must precede it due to runtime library [liblaser_geometry.so]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 1 is [/opt/ros/melodic/lib]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 2 is [/home/butterfly/ros1_bridge_ws/build/ros1_bridge]
    dir 3 is [/home/butterfly/ros_catkin_ws/devel/lib]`

Full report:

`Starting >>> ros1_bridge
[Processing: ros1_bridge]                                
--- stderr: ros1_bridge                                  
CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target dynamic_bridge
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:202 (custom_executable)

CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target dynamic_bridge
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/dashing/lib]
      dir 1 must precede it due to runtime library [liblaser_geometry.so]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 1 is [/opt/ros/melodic/lib]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 2 is [/home/butterfly/ros1_bridge_ws/build/ros1_bridge]
    dir 3 is [/home/butterfly/ros_catkin_ws/devel/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:202 (custom_executable)

CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target parameter_bridge
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:195 (custom_executable)

CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target parameter_bridge
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/dashing/lib]
      dir 1 must precede it due to runtime library [liblaser_geometry.so]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 1 is [/opt/ros/melodic/lib]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 2 is [/home/butterfly/ros1_bridge_ws/build/ros1_bridge]
    dir 3 is [/home/butterfly/ros_catkin_ws/devel/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:195 (custom_executable)

CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:188 (custom_executable)

CMake Warning at CMakeLists.txt:142 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/dashing/lib]
      dir 1 must precede it due to runtime library [liblaser_geometry.so]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 1 is [/opt/ros/melodic/lib]
      dir 3 must precede it due to runtime library [libimage_transport.so]
    dir 2 is [/home/butterfly/ros1_bridge_ws/build/ros1_bridge]
    dir 3 is [/home/butterfly/ros_catkin_ws/devel/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:188 (custom_executable)

CMake Warning at CMakeLists.txt:171 (add_library):
  Cannot generate a safe runtime search path for target ros1_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/aarch64-linux-gnu may be hidden by files in:
      /opt/ros/dashing/lib

  Some of these libraries may not be found correctly.

---
Finished <<< ros1_bridge [37.0s]

Summary: 1 package finished [38.1s]
  1 package had stderr output: ros1_bridge
`

#### Additional information
Please help
----

closed 2021-06-10 by larsvlent

gerni17, 2022-04-13:
Hi @larsvlent @Reuben-Lim .

How did you solve this issue?

ljaniec, 2022-06-23:
This may not be important, but the steps did not include `source` for both workspaces (`catkin_ws` for ROS1 and `ament_ws` for ROS 2). Could you check it?

```
source /opt/ros/noetic/setup.bash
source /opt/ros/foxy/setup.bash
source ~/catkin_ws/devel/setup.bash
source ~/ament_ws/install/local_setup.bash
colcon build --packages-select ros1_bridge --cmake-force-configure
```

faraz7321, 2023-02-05:
> Hi. I'm having your "old issue". How did you resolve it? My ROS1 is noetic while ROS2 is foxy.

any fix?

TheSeanParker, 2024-12-31:
Unfortunately by the ubuntu apt-get install the PCL 
```
sudo apt update
sudo apt install libpcl-dev
```
the PCL isntallled location is /usr/lib/x86_64-linux-gnu ;if you installed by yourself ,
```
cd PCL 
cmake ..
make 
sudo make install
```
the PCL lib will be installed in /usr/local/lib/
as we all know that: Linux(cover the ubuntu) find the lib the order is /usr/lib、/lib 和/usr/local/lib;
so your algorithm first find the PCL lib order is /usr/local/lib,the next is /usr/lib/x86_64-linux-gnu,so when CompileAndLink algorithm find the PCL in /usr/local/lib,but when run the algorithm exe,the ROS find PCL in /usr/lib/x86_64-linux-gnu,so the solution is delete lib in /usr/local/lib,
`sudo rm -r /usr/local/lib/libpcl*`
and you will not see the error of 
```
  Cannot generate a safe runtime search path for target fastlio_mapping
  because files in some directories may conflict with libraries in implicit
  directories:
    runtime library [libpcl_common.so.1.12] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /usr/local/lib
  Some of these libraries may not be found correctly.
```
and the interlinkage is my article ,hope you like and marked ,thanks
https://zhuanlan.zhihu.com/p/694151613