ros2/ros1_bridge #321: Build failed foxy+melodic:Failed to find ROS 2 package 'diagnostic_msgs'

javad123javad, 2021-08-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Melodic from APT. Foxy compiled from source
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
1. Foxy compiled according to ROS2-Foxy guide.
2. Foxy sourced from `~/ros2_foxy/install/setup.bash`
3. Melodic sourced from `/opt/ros/melodic/setup.bash`
4. colcon build started with:`colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure`

``` code that can be copy-pasted is preferred ``` -->
```
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
--- stderr: ros1_bridge                           
CMake Error at CMakeLists.txt:72 (message):
  Failed to find ROS 2 package 'diagnostic_msgs'

---
Failed   <<< ros1_bridge [25.6s, exited with code 1]

Summary: 0 packages finished [26.5s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge

```

clalancette, 2021-08-06:
I'll suggest you take a look at your `~/ros2_foxy/install` directory, and ensure that `diagnostic_msgs` appears in the list of packages there.  Also, after sourcing the Foxy `setup.bash`, take a look and ensure that `diagnostic_msgs` is in the `CMAKE_PREFIX_PATH`.

javad123javad, 2021-08-06:
> I'll suggest you take a look at your `~/ros2_foxy/install` directory, and ensure that `diagnostic_msgs` appears in the list of packages there. Also, after sourcing the Foxy `setup.bash`, take a look and ensure that `diagnostic_msgs` is in the `CMAKE_PREFIX_PATH`.

Thanks, Just checked the `~/ros2_foxy/install` directory. The `diagnostic_msgs` exists. When I execute `echo $CMAKE_PREFIX_PATH | grep diagnostic_msgs` , I can find `/home/imperiumdrive/ros2_foxy/install/diagnostic_msgs` in the variable.

clalancette, 2021-08-06:
Hm, I honestly don't know.  It works here when I try it out.

One thing I will say is that the docs explicitly say that you should source Melodic first, then Foxy.  I'd try that.  Otherwise I am currently out of ideas.

AskeyNil, 2021-11-12:
I have met the same problem, do you have a solution?

HuguesTHOMAS, 2021-11-12:
Hi, I am having the same issue here. I tried a bunch of different things to solve this without success.

Here:
https://github.com/ros2/ros1_bridge/blob/859e156aa351701442cdfe5507621b5bda86d1c8/CMakeLists.txt#L60-L61

I used `message()` to verify the path found by CMake

```
find_package(ament_lint_auto REQUIRED)
message("Found ament_lint_auto ${ament_lint_auto_DIR}")
find_package(diagnostic_msgs REQUIRED)
message("Found diagnostic_msgs ${diagnostic_msgs_DIR}")
```

It returns:

```
Found ament_lint_auto /opt/ros/foxy/install/ament_lint_auto/share/ament_lint_auto/cmake
Found diagnostic_msgs /opt/ros/melodic/share/diagnostic_msgs/cmake
```

Which means the wrong diagnostic message is found. So I modified with:

```
find_package(ament_lint_auto REQUIRED)
message("Found ament_lint_auto ${ament_lint_auto_DIR}")
# find_package(diagnostic_msgs REQUIRED)
find_package(diagnostic_msgs REQUIRED PATHS /opt/ros/foxy/install/diagnostic_msgs NO_DEFAULT_PATH)
message("Found diagnostic_msgs ${diagnostic_msgs_DIR}")
```

And now it returns

```
Found ament_lint_auto /opt/ros/foxy/install/ament_lint_auto/share/ament_lint_auto/cmake
Found diagnostic_msgs /opt/ros/foxy/install/diagnostic_msgs/share/diagnostic_msgs/cmake
```

Which is good. But the build still fails this time without any error message:

```
ROS_DISTRO was set to 'foxy' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'melodic' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'foxy' before. Please make sure that the environment does not mix paths from different distributions.
Starting >>> ros1_bridge
--- stderr: ros1_bridge
Found ament_lint_auto /opt/ros/foxy/install/ament_lint_auto/share/ament_lint_auto/cmake
Found diagnostic_msgs /opt/ros/foxy/install/diagnostic_msgs/share/diagnostic_msgs/cmake
---
Failed   <<< ros1_bridge [13.5s, exited with code -11]

Summary: 0 packages finished [14.1s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

The exit code is different now, so I don't know what to think

HuguesTHOMAS, 2021-11-12:
I eventually solved this issue by changing the order of the source commands. the README says:

```
# You have already sourced your ROS installation.
# Source your ROS 2 installation:
. <install-space-with-ros2>/local_setup.bash
# And if you have a ROS 1 overlay workspace, something like:
# . <install-space-to-ros1-overlay-ws>/setup.bash
# And if you have a ROS 2 overlay workspace, something like:
# . <install-space-to-ros2-overlay-ws>/local_setup.bash
```

However, I guess because the ROS2 installation is from source, it should be sourced after any ROS1 overlay. So here is the order I used for sourcing:

```
# Source your ROS1  instalation
source "/opt/ros/melodic/setup.bash"

# And if you have a ROS 1 overlay workspace, something like:
. "<path_to_catkin_ws>/install_isolated/setup.bash"

# Source your ROS 2 installation:
. "/opt/ros/foxy/install/setup.bash"

# And if you have a ROS 2 overlay workspace, something like:
. "<path_to_ros2_ws>/install/local_setup.bash"
```

gustavojoseleite, 2022-01-07:
Hi,

I tried the solution of @HuguesTHOMAS but doesn't work here (Same setup ROS2 Foxy + ROS1 Melodic) . Any other solution?

Thanks

gustavojoseleite, 2022-01-07:
Hi,

I tried the solution of @HuguesTHOMAS and worked!
But at the end of the build i receive another error:

`libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
`
Someone have the same error? How i could fix this?

Thanks!

lmendyk, 2022-02-21:
I have similar issue:

`/usr/bin/ld: libros1_bridge.so: undefined reference to ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_2_to_1(controller_manager_msgs::msg::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::ControllerState_<std::allocator<void> >&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_1_to_2(controller_manager_msgs::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::msg::ControllerState_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
`

gbiggs, 2022-02-22:
Are you using Foxy and Melodic on Ubuntu 18.04? I think it's worth noting that Foxy is not supported on 18.04 so errors are not surprising.

quarkytale, 2022-09-30:
**TD;LR The answer is not keep re-sourcing/re-overlaying in the same terminal.**

So this is happening with ROS1 Noetic + ROS2 Humble on Ubuntu 22.04 (Jammy) as well.
```
root@eb1dc6b69d3f:~/bridge_ws# source ~/ros2_humble_ws/install/setup.bash &&  source ~/ros2_msgs_ws/install/setup.bash &&  echo "add_compile_definitions(BOOST_BIND_GLOBAL_PLACEHOLDERS)" >> ~/bridge_ws/src/ros1_bridge/CMakeLists.txt &&  colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
ROS_DISTRO was set to 'humble' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.
Starting >>> ros1_bridge
--- stderr: ros1_bridge                         
CMake Error at CMakeLists.txt:79 (message):
  Failed to find ROS 2 package 'diagnostic_msgs'
---
Failed   <<< ros1_bridge [4.03s, exited with code 1]

Summary: 0 packages finished [4.56s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

Refer the figure:

[image]

1  |  2
3  |  4
Terminal 1 and 3 are the bridge_ws, 2 is ros1 only and 4 is ros2 related builds only,
Once you've built the respected ros workspaces along with msgs/srvs you want to map, open a fresh terminal for the bridge, overlay ros1 first, then ros2 and build the bridge (as @HuguesTHOMAS mentioned before). 

Suppose the bridge is still under development, and you clean the previous build. That's when it can fail to find the ros2 version of `diagnostic_msgs` despite unsetting the cache, https://github.com/ros2/ros1_bridge/blob/master/CMakeLists.txt#L78.

To avoid this, I found discarding that terminal and overlaying the workspaces in the correct order, in a yet again fresh terminal, seems to be fixing it. 
Though not a permanent solution, hope it helps!

kenloumixx, 2023-05-16:
@gustavojoseleite Hello, How did you solve [the latest error](https://github.com/ros2/ros1_bridge/issues/321#issuecomment-1007656985)? I am struggling with this for a long time so I'd want to ask if it is solved :)

nacayu, 2023-08-17:
> I eventually solved this issue by changing the order of the source commands. the README says:
> 
> ```
> # You have already sourced your ROS installation.
> # Source your ROS 2 installation:
> . <install-space-with-ros2>/local_setup.bash
> # And if you have a ROS 1 overlay workspace, something like:
> # . <install-space-to-ros1-overlay-ws>/setup.bash
> # And if you have a ROS 2 overlay workspace, something like:
> # . <install-space-to-ros2-overlay-ws>/local_setup.bash
> ```
> 
> However, I guess because the ROS2 installation is from source, it should be sourced after any ROS1 overlay. So here is the order I used for sourcing:
> 
> ```
> # Source your ROS1  instalation
> source "/opt/ros/melodic/setup.bash"
> 
> # And if you have a ROS 1 overlay workspace, something like:
> . "<path_to_catkin_ws>/install_isolated/setup.bash"
> 
> # Source your ROS 2 installation:
> . "/opt/ros/foxy/install/setup.bash"
> 
> # And if you have a ROS 2 overlay workspace, something like:
> . "<path_to_ros2_ws>/install/local_setup.bash"
> ```
> 
> Should the README be modified, or is this case too specific?
> 
> EDIT: And by the way, after doing this, I could use the original CMake file without the modifications I talked about in the last comment.

Thanks for your solution, I solved.

yuanliujia, 2024-11-19:
I found the same problem. In the morning, I woke up and reopened the terminal to try to build. I found no problem. But when I modified the msg file and rebuilt, it would appear.
`--- stderr: ros1_bridge
CMake Error at CMakeLists.txt:79 (message):
Failed to find ROS 2 package 'diagnostic_msgs'
---`
Finally, I found that I closed the original terminal for building and opened a new terminal. If you are not sure, you can execute in the new terminal
`unset CMAKE_PREFIX_PATH
unset ROS_PACKAGE_PATH
unset ROS_PATH
unset AMENT_PREFIX_PATH`
Then you don't need to source too much. Just source ros1_ws/install/setup.bash and ros2_ws/install/setup.bash where the message is located