ros2/ros1_bridge #329: Building ROS bridge issue

xander-m2k, 2021-09-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Foxy (ROS2) and Noetic (ROS1)
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
Install both debian packages for Noetic and Foxy:
http://wiki.ros.org/noetic/Installation/Ubuntu and https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html
Download the https://github.com/iirob/ros_opcua_communication repo and build a ROS1 workspace. (just catkin_make)
Download the https://github.com/TechmanRobotInc/tmr_ros2 repo and build a ROS2 workspace. (just colcon build)

Use the ros1-bridge installation instructions (this repository) to source both environments and build the ros1 bridge.

#### Expected behavior
Should compile
#### Actual behavior
```
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_1_to_2(controller_manager_msgs::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::msg::ControllerState_<std::allocator<void> >&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_2_to_1(controller_manager_msgs::msg::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::ControllerState_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/parameter_bridge.dir/build.make:368: parameter_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:252: CMakeFiles/parameter_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_1_to_2(controller_manager_msgs::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::msg::ControllerState_<std::allocator<void> >&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_2_to_1(controller_manager_msgs::msg::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::ControllerState_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/static_bridge.dir/build.make:368: static_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:198: CMakeFiles/static_bridge.dir/all] Error 2
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_1_to_2(controller_manager_msgs::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::msg::ControllerState_<std::allocator<void> >&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_2_to_1(controller_manager_msgs::msg::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::ControllerState_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dynamic_bridge.dir/build.make:368: dynamic_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:306: CMakeFiles/dynamic_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
#### Additional information
I also tried excluding the tm_robot repo from the build, but I got the same error. The OPCUA library built before, so that should not be the problem.

xander-m2k, 2021-09-22:
Reinstalling both ROS distros solved the problem

closed 2021-09-22 by xander-m2k

likemountain, 2022-03-25:
Hello,
I am having the same error when building the ros1_bridge. @LoyVanBeek How did you remove and re-install `controller_manager_msgs`? Could you explain more detailed procedures to do this?

LoyVanBeek, 2022-03-27:
It's embarrassingly simple and not a real fix at all: 

1. `sudo apt remove ros-noetic-controller-manager-msgs`
2. Copy & paste all the pkgs that are now going to be uninstalled because they depend on the ros-noetic-controller-manager-msgs :cry: 
3. Build ros1_bridge as per instructions
4. `sudo apt install ros-noetic-controller-manager-msgs` + the pkgs remembered from step 2. 
That is all there is to it.

LoyVanBeek, 2022-04-28:
Alternatively, you can (of course, in hindsight) remove `sudo apt remove ros-foxy-controller-manager-msgs` and leave the `noetic` version in place. Which is better depens on your setup and which package has more dependents. 

Still a workaround and not a real fix from me unfortunately.

awied1404, 2022-08-12:
I had controller_manager_msgs installed in both ROS Distros. For me the solution was to just ignore the error because the binaries were built (don't know exactly why) and I could go on with the tutorials like described in the README and they all worked for me

LoyVanBeek, 2022-10-25:
I think https://github.com/ros2/ros1_bridge/pull/382 is related to this problem. Not entirely because my PR does not fix this issue here, but the solution is somewhere around there...