ros2/ros1_bridge #343: Example 4 of the README.md is failing

ckaran, 2022-02-09:
## Bug report

Example 4 of the README.md is failing.

**Required Info:**

- Operating System:

```bash
lsb_release -a:
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

uname -a:
Linux Hammer 5.13.0-28-generic #31~20.04.1-Ubuntu SMP Wed Jan 19 14:08:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

- Installation type:

Both binaries and source are installed.  I have `noetic` and `galactic` installed as binaries, and have the `galactic` sources so that I can build the bridge as the instructions in the README.md file say I should.

- Version or commit hash:

83390800c5ff205625428f148b514008932cfc61 within the repo.  Within `ros2.repos`, the relevant lines are:

```yaml
  ros2/ros1_bridge:
    type: git
    url: https://github.com/ros2/ros1_bridge.git
    version: galactic
```

- DDS implementation:

I don't know how to determine this.  I haven't made any modifications to the installations, so whatever the default is.

- Client library (if applicable):

N/A

#### Steps to reproduce issue

1. Ensure that your `.bashrc` doesn't source any flavor of ROS by default.
2. Follow the instructions in the prerequisites section of the README.md at https://github.com/ros2/ros1_bridge#prerequisites.
3. Build the bridge as described at https://github.com/ros2/ros1_bridge#building-the-bridge-from-source.
4. Skip down to example 4 at https://github.com/ros2/ros1_bridge#example-4-bridge-only-selected-topics-and-services and do as it says.

#### Expected behavior

I expected to see the talker and listener working in their respective terminals.  This worked correctly.  I also expected the following to work:

```bash
# Shell F:
. <install-space-with-ros2>/setup.bash
ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{a: 1, b: 2}"
```

#### Actual behavior

Several errors occurred.  First, the `bridge.yaml` file appeared to be missing a ros2 service:

```yaml
topics:
  -
    topic: /chatter  # ROS1 topic name
    type: std_msgs/msg/String  # ROS2 type name
    queue_size: 1  # For the publisher back to ROS1
services_1_to_2:
  -
    service: /add_two_ints  # ROS1 service name
    type: example_interfaces/srv/AddTwoInts  # The ROS2 type name
```

This caused a failure on [lines 230-233](https://github.com/ros2/ros1_bridge/blob/fe94cf5b9bd3987e00c3ebf02c49474ce78d0a1c/src/parameter_bridge.cpp#L230-L233) because there isn't a service available.  I commented out [lines 174-234](https://github.com/ros2/ros1_bridge/blob/fe94cf5b9bd3987e00c3ebf02c49474ce78d0a1c/src/parameter_bridge.cpp#L174-L234), recompiled and reran the tests.  This allowed me to get past the error, but it still didn't make the service work; the `ros2` service call hung forever waiting for the service to be available, and the server waited to get called.   I'm now at a loss as to what I'm supposed to do next.

gbiggs, 2022-02-21:
I was successfully able to follow the README and run all examples, including example 4. Please ask for help at ROS Answers, and be sure to post the exact error output you receive.

closed 2022-02-21 by gbiggs

ckaran, 2022-02-22:
@gbiggs Did you rebuild from a completely clean pull of ROS 1 and ROS 2?

gbiggs, 2022-02-22:
Yes, I did.

ckaran, 2022-02-23:
OK, I'll poke at it again on my end and try to figure out what's going on.  Thank you!