ros2/ros1_bridge #344: Building ros1_bridge with vision_msgs installed (Melodic + Eloquent) fails

asasine, 2022-02-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - b5e4931cefce6cc3351b31b04b0399a6d3bacac4
- DDS implementation:
  - Fast-RTPS (Eloquent default)
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
After installing ROS1 Melodic and ROS2 Eloquent, and installing ros-melodic-vision-msgs and ros-eloquent-vision-msgs,
 building ros1_bridge fails with the following error:

```
FAILED: CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp.o 
/usr/bin/c++  -Dros1_bridge_EXPORTS -I/home/adam/src/Azure-Compute-Robotics/src/robot/bridge/ros1_bridge/include -Igenerated -I/home/adam/src/Azure-Compute-Robotics/install/ros2/nvapriltags_ros2/include -I/home/adam/src/Azure-Compute-Robotics/install/ros2/azc_msgs/include -I/opt/ros/eloquent/include -I/opt/ros/melodic/include -I/home/adam/src/Azure-Compute-Robotics/install/ros1/azc_msgs/include -I/usr/include/eigen3 -I/home/adam/src/Azure-Compute-Robotics/install/ros1/realsense2_camera/include -I/usr/include/opencv -I/home/adam/src/Azure-Compute-Robotics/install/ros1/ros_sdr_agent_msgs/include -I/usr/include/suitesparse -I/opt/ros/melodic/include/g2o -I/home/adam/src/Azure-Compute-Robotics/install/ros1/wibotic_msg/include -I/usr/include/OGRE/Overlay -I/usr/include/OGRE -g -fPIC   -Wall -Wextra -std=gnu++14 -Wno-unused-parameter -MD -MT CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp.o -MF CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp.o.d -o CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp.o -c generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp
generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_2_to_1(const ROS2_T&, ROS1_T&) [with ROS1_T = vision_msgs::ObjectHypothesisWithPose_<std::allocator<void> >; ROS2_T = vision_msgs::msg::ObjectHypothesisWithPose_<std::allocator<void> >]’:
generated/vision_msgs__msg__ObjectHypothesisWithPose__factories.cpp:87:26: error: cannot convert ‘const _id_type {aka const std::__cxx11::basic_string<char>}’ to ‘vision_msgs::ObjectHypothesisWithPose_<std::allocator<void> >::_id_type {aka long int}’ in assignment
   ros1_msg.id = ros2_msg.id;
                          ^~

[271/297] Building CXX object CMakeFiles/ros1_bridge.dir/g...erated/vision_msgs__msg__ObjectHypothesis__factories.cpp.o[K
FAILED: CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp.o 
/usr/bin/c++  -Dros1_bridge_EXPORTS -I/home/adam/src/Azure-Compute-Robotics/src/robot/bridge/ros1_bridge/include -Igenerated -I/home/adam/src/Azure-Compute-Robotics/install/ros2/nvapriltags_ros2/include -I/home/adam/src/Azure-Compute-Robotics/install/ros2/azc_msgs/include -I/opt/ros/eloquent/include -I/opt/ros/melodic/include -I/home/adam/src/Azure-Compute-Robotics/install/ros1/azc_msgs/include -I/usr/include/eigen3 -I/home/adam/src/Azure-Compute-Robotics/install/ros1/realsense2_camera/include -I/usr/include/opencv -I/home/adam/src/Azure-Compute-Robotics/install/ros1/ros_sdr_agent_msgs/include -I/usr/include/suitesparse -I/opt/ros/melodic/include/g2o -I/home/adam/src/Azure-Compute-Robotics/install/ros1/wibotic_msg/include -I/usr/include/OGRE/Overlay -I/usr/include/OGRE -g -fPIC   -Wall -Wextra -std=gnu++14 -Wno-unused-parameter -MD -MT CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp.o -MF CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp.o.d -o CMakeFiles/ros1_bridge.dir/generated/vision_msgs__msg__ObjectHypothesis__factories.cpp.o -c generated/vision_msgs__msg__ObjectHypothesis__factories.cpp
generated/vision_msgs__msg__ObjectHypothesis__factories.cpp: In static member function ‘static void ros1_bridge::Factory<ROS1_T, ROS2_T>::convert_2_to_1(const ROS2_T&, ROS1_T&) [with ROS1_T = vision_msgs::ObjectHypothesis_<std::allocator<void> >; ROS2_T = vision_msgs::msg::ObjectHypothesis_<std::allocator<void> >]’:
generated/vision_msgs__msg__ObjectHypothesis__factories.cpp:79:26: error: cannot convert ‘const _id_type {aka const std::__cxx11::basic_string<char>}’ to ‘vision_msgs::ObjectHypothesis_<std::allocator<void> >::_id_type {aka long int}’ in assignment
   ros1_msg.id = ros2_msg.id;
                          ^~
```

#### Expected behavior
ros1_bridge should successfully build by not associating ROS1 and ROS2 versions of vision_msgs/msg/ObjectHypothesisWithPose and vision_msgs/msg/ObjectHypothesis, as the types for the field `id` do not match.

#### Actual behavior
ros1_bridge build fails since it tries to associate ROS1 and ROS2 versions of vision_msgs/msg/ObjectHypothesisWithPose and vision_msgs/msg/ObjectHypothesis, despite the types for the field `id` not matching.

#### Additional information
ROS1 vision_msgs/msg/ObjectHypothesis uses an `int64` for the `id` field. ROS2 vision_msgs/msg/ObjectHypothesis uses a `string` for the `id` field.

The default field matching rules, according to the [docs in the repo](https://github.com/ros2/ros1_bridge/blob/fe94cf5b9bd3987e00c3ebf02c49474ce78d0a1c/doc/index.rst#how-are-ros-1-and-2-messages-associated-with-each-other), indicate automatic message associations are only created when all field types match.

Related discussion in vision_msgs: https://github.com/ros-perception/vision_msgs/issues/39

gbiggs, 2022-02-19:
ROS 2 Eloquent is no longer supported. Does this problem occur on a more recent version of ROS 2?

clalancette, 2022-03-10:
No response in over two weeks, so closing.  Feel free to respond to the last question and reopen.

closed 2022-03-10 by clalancette

cuge1995, 2022-10-04:
sudo apt-get install ros-melodic-vision-msgs