ros2/ros1_bridge #348: Failed to build ros1_bridge from source when srv includes msg in a same package

nqduy35, 2022-03-24:
Trying to build a two very basic examples of using ros1_bridge: **test_one** and **test_two**. The content of the two examples are almost the same except that the one srv file of the second one includes the type of a msg. I can build the ros1_bridge from souce in the first example, but not in the second example. Following is the detail: 

In **test_one**, there are two msg and one srv: 
msg/Data.msg: 
int32 a

msg/Info.msg
Data data

srv/Print.srv
int32 input
---
string output

Exacly the same with the above, the only difference of the **test_two** is the srv's content: 

srv/Print.srv
> Data input
> ---
> string output

I also tried to change the the first line into 
> test_two/Data input 
but the error is still. 

## Bug report
- Operating System: _Ubuntu 20.04, ROS noetic and ROS2 foxy_
- Installation type: _from source_
- Version or commit hash: 3209bc77c77b072013b05bcc6690a24a6bd16aeb

#### Steps to reproduce issue

> $ source /opt/ros/noetic/setup.bash
> $ source /opt/ros/foxy/setup.bash
> $ source ~/catkin_ws/devel/setup.bash
> $ source ~/ament_ws/install/local_setup.bash
> $ colcon build --package-select ros1_bridge --cmake-force-configure

#### Actual error
> --- stderr: ros1_bridge                                          
> /usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<test_two::Data_<std::allocator<void> >, test_two::msg::Data_<std::allocator<void> > >::convert_1_to_2(test_two::Data_<std::allocator<void> > const&, test_two::msg::Data_<std::allocator<void> >&)'
> /usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<test_two::Data_<std::allocator<void> >, test_two::msg::Data_<std::allocator<void> > >::convert_2_to_1(test_two::msg::Data_<std::allocator<void> > const&, test_two::Data_<std::allocator<void> >&)'
> collect2: error: ld returned 1 exit status
> make[2]: *** [CMakeFiles/static_bridge.dir/build.make:368: static_bridge] Error 1
> make[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/static_bridge.dir/all] Error 2
> make[1]: *** Waiting for unfinished jobs....
> /usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<test_two::Data_<std::allocator<void> >, test_two::msg::Data_<std::allocator<void> > >::convert_1_to_2(test_two::Data_<std::allocator<void> > const&, test_two::msg::Data_<std::allocator<void> >&)'
> /usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<test_two::Data_<std::allocator<void> >, test_two::msg::Data_<std::allocator<void> > >::convert_2_to_1(test_two::msg::Data_<std::allocator<void> > const&, test_two::Data_<std::allocator<void> >&)'
> collect2: error: ld returned 1 exit status
> make[2]: *** [CMakeFiles/parameter_bridge.dir/build.make:368: parameter_bridge] Error 1
> make[1]: *** [CMakeFiles/Makefile2:314: CMakeFiles/parameter_bridge.dir/all] Error 2
> /usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<test_two::Data_<std::allocator<void> >, test_two::msg::Data_<std::allocator<void> > >::convert_1_to_2(test_two::Data_<std::allocator<void> > const&, test_two::msg::Data_<std::allocator<void> >&)'
> /usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<test_two::Data_<std::allocator<void> >, test_two::msg::Data_<std::allocator<void> > >::convert_2_to_1(test_two::msg::Data_<std::allocator<void> > const&, test_two::Data_<std::allocator<void> >&)'
> collect2: error: ld returned 1 exit status
> make[2]: *** [CMakeFiles/dynamic_bridge.dir/build.make:368: dynamic_bridge] Error 1
> make[1]: *** [CMakeFiles/Makefile2:341: CMakeFiles/dynamic_bridge.dir/all] Error 2
> make: *** [Makefile:141: all] Error 2
> ---
> Failed   <<< ros1_bridge [4min 26s, exited with code 2]

gbiggs, 2022-03-24:
Can you please provide a repository or tarball containing the working and failing cases that I can use to reproduce the error?

nqduy35, 2022-03-24:
Thank your for your answer. The code is available in here git@github.com:nqduy35/ros1_bridge_348.git

amilcarlucas, 2022-03-25:
I get the same typo of linking errors when using `rmw_fastrtps_cpp` on galactic.

I get no linking errors when not using `rmw_fastrtps_cpp` .

@pablogs9 do I need to install `fastrtpsgen` or something like that?

amilcarlucas, 2022-03-28:
I seam to have fixed the linking error by installing `ros-galactic-rmw-fastrtps-dynamic-cpp`

@pablogs9  you might what to add that to the documentation.

nqduy35, 2022-03-28:
Literally, the problem I raised is not the same to the problem of @amilcarlucas. The issue I raised is illustrated by a very basic example at [github.com:nqduy35/ros1_bridge_348](https://github.com/nqduy35/ros1_bridge_348). 
@gbiggs Please tell me if you need other information. Thank you.

gbiggs, 2022-03-28:
Not as yet. I will take a look at your example and try to replicate the problem. If I can replicate it, then we will work on fixing it.

nqduy35, 2022-04-08:
@gbiggs Sorry if I disturb you too much. I just wanted to know if you have already tried to reproduce the error? Our team wants to be sure that it is not the fault of my side so they can choose another solution. By the way, I have added two new images [github.com:nqduy35/ros1_bridge_348](https://github.com/nqduy35/ros1_bridge_348) to see the problem better.

amilcarlucas, 2022-04-11:
I have two messages and three services on the same package. And I got it to run now, after updating to the latest micro_ros_arduino version. This version uses the latest rmw version.

gbiggs, 2022-05-05:
@nqduy35 I've been able to reproduce the error. I'll start figuring out the cause and a fix.

gbiggs, 2022-06-23:
@nqduy35 Just to let you know I'm still working on this, but have not made progress in identifying the error yet.

gbiggs, 2022-07-05:
I found some more time to work on this, and managed to identify the source of your problem.

In ROS 2, interface definitions (messages, services and actions) are, by convention, stored in their own package that has `_interfaces` (new) or `_msgs` as the suffix of its name. This is only a convention: in general things will work just fine if you don't do this. However the `ros1_bridge` has this convention hard-coded in when it searches for matching interface definitions between ROS 1 and ROS 2. If you rename your `test_two` package to `test_two_interfaces`, the bridge will compile.

closed 2022-07-05 by gbiggs

nqduy35, 2022-07-06:
@gbiggs Thank you very much for your time and effort.