ros2/ros1_bridge #349: ROS1 Bridge doesn't work with custom service

Fraunhofer0126, 2022-03-24:
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - ros1: noetic, ros2:galactic

#### Steps
I have 2 workspaces.

In the ros2 workspace, I have `cpp_srvcli` and `ex_interfaces` package. In `cpp_srvcli`, I just copy `add_two_ints_client.cpp` and `add_two_ints_server.cpp` from ros2 wiki; In `ex_interfaces`, I define a srv named `AddTwoInts.srv`
```
int64 a
int64 b
---
int64 sum
```
In the ros1 workspace, I have one package `roscpp_kakaka` to which just copy `roscpp_tutorials` package from `https://github.com/ros/ros_tutorials/tree/noetic-devel/roscpp_tutorials` and change its name. And I use the srv named `TwoInts.srv`
```
int64 a
int64 b
---
int64 sum
```
What's more, I followed `https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst` and added a `my_mapping_rules.yaml` in the package `ex_interfaces`. I checked the `rosservice list` and `ros2 service list`, the services names are both `/add_two_ints`.
```
-
  ros1_package_name: 'roscpp_kakaka'
  ros1_service_name: '/add_two_ints'
  ros2_package_name: 'ex_interfaces'
  ros2_service_name: '/add_two_ints'
```
After I built the ros1_bridge and run 
```
# Shell A:
roscore -p 11311
```
```
# Shell B:
. <ros-install-dir>/setup.bash
. <ros2-install-dir>/setup.bash
export ROS_MASTER_URI=http://localhost:11311
ros2 run ros1_bridge dynamic_bridge
```
```
# Shell C:
. <ros-install-dir>/setup.bash
export ROS_MASTER_URI=http://localhost:11311
rosrun roscpp_kakaka add_two_ints_server
```
```
# Shell D:
. <ros2-install-dir>/setup.bash
ros2 run cpp_srvcli client x y
```
in seperate terminals, the Shell D response
```
[INFO] [1648132685.709708988] [rclcpp]: service not available, waiting again...
```

Could you help me? The ros2 client and ros2 server works; And the ros1 client and ros1 server works, too.

Surely, I modified the `add_two_ints_client.cpp" in ros2 package to fit my package name `ex_interface` instead of `example_interfaces`.

I also run `ros2 run ros1_bridge dynamic_bridge --print-pairs` to check pairs, but there are only these pairs like
```
- 'example_interfaces/srv/AddTwoInts' (ROS 2) <=> 'roscpp_tutorials/TwoInts' (ROS 1)
- 'example_interfaces/srv/AddTwoInts' (ROS 2) <=> 'rospy_tutorials/AddTwoInts' (ROS 1)
```
I can't find any pair about my package `ex_interfaces` or `cpp_srvcli`.

I'm a noob and need help QAQ.

poplinis, 2022-05-03:
Where are you building `ros1_bridge`? Do you source its install location after building? I was having a very similar issue that was caused by accidentally running the prebuilt `ros1_bridge` that came with my ros2 install (and thus only included the default interfaces) rather than the version that was built to include my custom messages.

closed 2022-05-03 by Fraunhofer0126

AlejandroGarcia03, 2024-10-17:
Wait, so you have a way to bridge also custom messages? I've been trying to do that, I've also tried sourcing the install location where I had my custom messages, was there anything else you tried for doing so?