ros2/ros1_bridge #352: sensor_msgs::PointCloud2 message freeze

Thomas-Merlet, 2022-04-01:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04 LTS
- Installation type:
  -  Installed using binaries `sudo apt install ros-galactic-ros1-bridge`
- Version or commit hash:
  -  0.10.1-2focal.20220211.121358
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

I've tried using ROS 1 Noetic and ROS 2 Dashing and Galactic. 

1. Play a bag file with sensor_msgs::PointCloud2 data in ROS1
2. Launch the bridge using `ros2 run ros1_bridge dynamic_bridge`
3. Launch multiple ROS1 nodes (like Autoware.IA ROS 1 nodes for example)
4. Launch a ROS2 nodes that subscribe to the raw sensor_msgs::PointCloud2 rosbag data

#### Expected behavior

No freeze in sensor_msgs::PointCloud2 messages.

#### Actual behavior

sensor_msgs::PointCloud2 messages freeze about every 2 seconds as shown below:

https://user-images.githubusercontent.com/26826861/161289762-5aa43ca7-a7a0-46cc-abda-54bf9bd40bd1.mp4

#### Additional information

This seems to happen when multiple ROS 1 nodes are running. Stopping the ROS 1 nodes solve the issue. 

`top` gives the following output:
[image]

gbiggs, 2022-04-01:
Can you please provide a reproducible example that does not depend on such a large project? That will make it much easier to debug.

Thomas-Merlet, 2022-07-22:
Hi,
To follow up on this issue, we were able to resolve the problem by specifying the requested topics, using `parameter_bridge`.

I'm unable to produce a small reproducable example when using the `dynamic_bridge`. Although the number of topics over the bridge are approximatly the same, maybe the lower queue size (value of 1 in our `bridge.yaml` file) might explain the different behaviour we are getting.

gbiggs, 2022-07-24:
Thanks for the update. I'll close this for now, then.

closed 2022-07-24 by gbiggs