ros2/ros1_bridge #353: Fields mapping

amifsud, 2022-04-08:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04

- Installation type: Binaries, ros1_bridge from sources
- Version or commit hash: Foxy HEAD
- DDS implementation: Compile time issue

#### Steps to reproduce issue

Trying to map the following message in ros1:

Example.msg :
```
float64 a
float64 b
float64 c
float64 d
```
to this one in ros2:

Example1.msg :
```
float64 w
float64 x
float64 y
float64 z
```

With the mapping rules :

```
-
    ros1_package_name: 'ros1_msgs'
    ros1_message_name: 'Example'
    ros2_package_name: 'ros2_msgs'
    ros2_message_name: 'Example1'
        fields_1_to_2:
            a: 'w'
            b: 'x'
            c: 'y'
            d: 'z'
```

Or the following ones :
```
-
    ros1_package_name: 'ros1_msgs'
    ros1_message_name: 'Example'
    ros2_package_name: 'ros2_msgs'
    ros2_message_name: 'Example1'
        fields_2_to_1:
            w: 'a'
            x: 'b'
            y: 'c'
            z: 'd'
```

In ros1 workspace (~/ros1_ws):

```
source /opt/ros/noetic/setup.bash
catkin_make install

source ~/ros1_ws/install/setup.bash
rosmsg show ros1_msgs/Example
```

In ros2 workspace (~/ros2_ws):

```
source /opt/ros/foxy/setup.bash
colcon build --build-base foxy_build --install-base foxy_install

source ~/ros2_ws/foxy_install/setup.bash
ros2 interface show ros2_msgs/msg/Example1
```

In ros1_bridge workspace (~/bridge_ws):

```
source ~/ros1_ws/install/setup.bash
source ~/ros2_ws/foxy_install/setup.bash
colcon build --build-base foxy_build --install-base foxy_install --cmake-force-configure

source ~/bridge_ws/foxy_install/setup.bash
ros2 run ros1_bridge dynamic_bridge --print-pairs | grep Example
```

#### Expected behavior

In ros1 workspace (~/ros1_ws):

```
source /opt/ros/noetic/setup.bash
catkin_make install

source ~/ros1_ws/install/setup.bash
rosmsg show ros1_msgs/Example
```
**Result :** 
```
float64 a
float64 b
float64 c
float64 d
```

In ros2 workspace (~/ros2_ws):

```
source /opt/ros/foxy/setup.bash
colcon build --build-base foxy_build --install-base foxy_install

source ~/ros2_ws/foxy_install/setup.bash
ros2 interface show ros2_msgs/msg/Example1
```
**Result :**
```
float64 w
float64 x
float64 y
float64 z
```

In ros1_bridge workspace (~/bridge_ws):

```
source ~/ros1_ws/install/setup.bash
source ~/ros2_ws/foxy_install/setup.bash
colcon build --build-base foxy_build --install-base foxy_install --cmake-force-configure
```
**Result :** No error or warning for both mapping rules

```
source ~/bridge_ws/foxy_install/setup.bash
ros2 run ros1_bridge dynamic_bridge --print-pairs | grep Example
```

**Result :** for both mapping rules :
```
 - 'icars2_msgs/msg/Example1' (ROS 2) <=> 'icars_msgs/Example' (ROS 1)
 ```

#### Actual behavior

Expected behavior until in ros1_bridge workspace.

In ros1_bridge workspace (~/bridge_ws):

```
source ~/ros1_ws/install/setup.bash
source ~/ros2_ws/foxy_install/setup.bash
colcon build --build-base foxy_build --install-base foxy_install --cmake-force-configure
```
**Result:** 
For the second mapping rule (using field_2_to_1)
```
--- stderr: ros1_bridge                                          
Mapping for package icars2_msgs contains unknown field(s)
---
```
Expected behavior for the first.

```
source ~/bridge_ws/foxy_install/setup.bash
ros2 run ros1_bridge dynamic_bridge --print-pairs | grep Example
```
**Result:** Empty answer for the second mapping rule, expected behavior for the first.

gbiggs, 2022-05-05:
Can you please provide a copy of your source to reproduce the issue?

amifsud, 2022-05-05:
[Here](https://github.com/amifsud/fields_mapping.git) is a repo with the different workspaces (ros1_bridge as a submodule)

Thanks again for your reply
Alexis

quarkytale, 2022-05-21:
Thanks for sharing the source code. I could reproduce both #353 and [#354](https://github.com/ros2/ros1_bridge/issues/354). Looking into it.

gbiggs, 2022-05-28:
@amifsud Please see [this comment](https://github.com/ros2/ros1_bridge/issues/354#issuecomment-1140135615) on #354. I believe this issue has the same cause. Try running the commands in three separate terminals.

amifsud, 2022-05-30:
Hello,

As I said in #354, I actually used different terminals.

Thanks,
Alexis

quarkytale, 2022-06-02:
@amifsud I think this might solve your issue, please try again using [quarkytale/mapping_rule](https://github.com/ros2/ros1_bridge/tree/quarkytale/mapping_rule) branch of `ros1_bridge`. It should print out using both mapping rules. Let me know how it goes!

amifsud, 2022-06-08:
@quarkytale Yes it is working for both mapping rules. Thanks !

quarkytale, 2022-06-08:
Thanks! I'll open a PR for it and update here once its merged.

closed 2022-06-27 by quarkytale