ros2/ros1_bridge #356: demo ros2 call ros1 service failure

xlla, 2022-04-24:
## Bug report

**Required Info:**

- Operating System:
  - OSX 10.13
- Installation type:
  - from source
- Version or commit hash:
  - foxy branch mixed  some new branch and tags
- DDS implementation:
  - Fast RTPS
- Client library (if applicable):
  - 

#### Steps to reproduce issue

exactly as Example 4: bridge only selected topics and services

#### Expected behavior
as Example 4 described

#### Actual behavior

bridge create failure, can't call ros1 service from ros2

roscore was running on machine A

launch bridge on machine B
<pre>
$ ros2 run ros1_bridge parameter_bridge
Trying to create bidirectional bridge for topic '/chatter' with ROS 2 type 'std_msgs/msg/String'
[INFO] [1650822995.675139598] [ros_bridge]: create bidirectional bridge for topic /chatter
Trying to create bridge for ROS 2 service '/add_two_ints' with type 'example_interfaces/srv/AddTwoInts'
failed to create bridge ROS 1 service '/add_two_ints' no conversion for type 'example_interfaces/srv/AddTwoInts'
The parameter 'services_2_to_1' either doesn't exist or isn't an array
[INFO] [1650822996.207521866] [ros_bridge]: Passing message from ROS 1 std_msgs/String to ROS 2 std_msgs/msg/String (showing msg only once per type)
</pre>

launch ros1 service on machine C
<pre>
$ rosrun roscpp_tutorials add_two_ints_server
</pre>

launch ros2 service caller on machine B
<pre>
$ ros2 service call /add_two_ints example_interfaces/srv/AddTwoInts "{a: 1, b: 2}"
waiting for service to become available...
</pre>

ros1 service was launch 
<pre>
$ rosservice list
/add_two_ints
/add_two_ints_server/get_loggers
/add_two_ints_server/set_logger_level
/ros_bridge/get_loggers
/ros_bridge/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/talker_97489_1650826988152/get_loggers
/talker_97489_1650826988152/set_logger_level
/talker_97842_1650827739573/get_loggers
/talker_97842_1650827739573/set_logger_level
</pre>

ros1 service can be call from ros1 client
<pre>
$ rosrun roscpp_tutorials add_two_ints_client 55 33
[ INFO] [1650827853.334119000]: Sum: 88
</pre>

#### Additional information

## Feature request

#### Feature description

#### Implementation considerations

xlla, 2022-04-25:
I have created another ubuntu 20.04 server , build ros2 galactic from source, it still failure to run Example 4.

<pre>
$ ros2 run ros1_bridge parameter_bridge
Trying to create bidirectional bridge for topic '/chatter' with ROS 2 type 'std_msgs/msg/String'
[INFO] [1650894884.831267657] [ros_bridge]: create bidirectional bridge for topic /chatter
Trying to create bridge for ROS 2 service '/add_two_ints' with type 'example_interfaces/srv/AddTwoInts'
Created 1 to 2 bridge for service /add_two_ints
The parameter 'services_2_to_1' either doesn't exist or isn't an array
[INFO] [1650894885.151093541] [ros_bridge]: Passing message from ROS 1 std_msgs/String to ROS 2 std_msgs/msg/String (showing msg only once per type)
</pre>

xlla, 2022-05-02:
I unconsciously run ros1 client with ros2 server on this parameter_bridge, it is work!

So I think the bridge yaml for example 4 was wrong, the corrected should:

<pre>
topics:
  -
    topic: /chatter  # ROS1 topic name
    type: std_msgs/msg/String  # ROS2 type name
    queue_size: 1  # For the publisher back to ROS1
services_2_to_1:
  -
    service: /add_two_ints  # ROS1 service name
    type: example_interfaces/srv/AddTwoInts  # The ROS2 type name
</pre>

see the PR #357

methylDragon, 2022-05-19:
Thanks for catching this! I'll follow up on the PR (: