ros2/ros1_bridge #358: tf2 messages not forwarded through ros1_bridge

Mechaick, 2022-05-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20
- Installation type:
  - binairies

#### Steps to reproduce issue

My robot is composed of two computers, a raspberry pi running ubuntu 18 and sending the tf messages to a jetson nano. On the jetson nano there is a ubuntu 20 and the ROS1_bridge running

#### Expected behavior

I should be able to see the messages when doing ros2 topic echo /tf

#### Actual behavior

ros1_bridge is giving the following issue:

`failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair.`

gbiggs, 2022-05-05:
Can you please provide a [minimal reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example) so we can verify and fix this bug?

Mechaick, 2022-05-05:
Yes, Sadly I will only be available to provide it on next Tuesday. 
Just to be sure, do you only need the code I do to execute the ros1_bridge or also how I configurated the devices?

gbiggs, 2022-05-05:
Please provide as much information as you can.

Mechaick, 2022-05-10:
This is the code I do to have the issue:

Everything is run on a jetson nano using ubuntu 20.4.

_Terminal1_ 
`source /opt/ros/noetic/setup.bash`
`rostopic list`

```
[...]
/tf
/tf_static
```

_Terminal2_

`source /opt/ros/foxy/setup.bash`
`cd colcon_ws`
`source install/setup.bash`
`ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`

_Terminal 3_

`source /opt/ros/foxy/setup.bash`
`ros2 topic echo /tf tf2_msgs/TFMessage`

After this I got nothing but the following message on the terminal 2

_Terminal 2_

```
failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option

```

methylDragon, 2022-05-25:
If the ROS 1 tf publisher is on Ubuntu 20.04, I get no issues.

It might be related to 18.04 -> 20.04?
We need a reproducible example

[image]

Mechaick, 2022-06-02:
Issue Solved !

My issue was in fact terribly simple. The previous people installing the bridge downloaded it through git and compiled it without all the dependencies. I removed it and installed from apt. Now it's working, sorry for your lost time.

closed 2022-06-02 by Mechaick

kenloumixx, 2023-06-01:
@Mechaick  Hello, may I ask a question? I am suffering from the same issue, found your answer but for me it's not still clear.. I also downloaded the ros1_bridge through apt and you mean that installing it without apt matters?