ros2/ros1_bridge #359: Failed   <<< ros1_bridge [7min 54s, exited with code 2]

Baldins, 2022-05-03:
Hello, I am getting the following error building the bridge . 

Someone can help me solving it? 

francesca@system76-pc:~/ros1_bridge_ws$ colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
ERROR:colcon.colcon_core.package_selection:Exception in package selection extension 'override_check': argument --allow-overriding: conflicting option string: --allow-overriding
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/package_selection/__init__.py", line 109, in _add_package_selection_arguments
    retval = extension.add_arguments(parser=group)
  File "/usr/lib/python3/dist-packages/colcon_override_check/package_selection/override_check.py", line 52, in add_arguments
    parser.add_argument(
  File "/usr/lib/python3/dist-packages/colcon_defaults/argument_parser/defaults.py", line 78, in add_argument
    argument = super().add_argument(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/argument_parser/destination_collector.py", line 46, in add_argument
    argument = super().add_argument(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/argument_parser/type_collector.py", line 34, in add_argument
    ret = super().add_argument(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_core/argument_parser/__init__.py", line 171, in add_argument
    return self._parser.add_argument(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/colcon_argcomplete/argument_parser/argcomplete/__init__.py", line 49, in add_argument
    argument = self._parser.add_argument(*args, **kwargs)
  File "/usr/lib/python3.8/argparse.py", line 1398, in add_argument
    return self._add_action(action)
  File "/usr/lib/python3.8/argparse.py", line 1602, in _add_action
    action = super(_ArgumentGroup, self)._add_action(action)
  File "/usr/lib/python3.8/argparse.py", line 1412, in _add_action
    self._check_conflict(action)
  File "/usr/lib/python3.8/argparse.py", line 1551, in _check_conflict
    conflict_handler(action, confl_optionals)
  File "/usr/lib/python3.8/argparse.py", line 1560, in _handle_conflict_error
    raise ArgumentError(action, message % conflict_string)
argparse.ArgumentError: argument --allow-overriding: conflicting option string: --allow-overriding

Starting >>> ros1_bridge
[Processing: ros1_bridge]                             
[Processing: ros1_bridge]                                     
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                        
[Processing: ros1_bridge]                                        
[Processing: ros1_bridge]                                        
--- stderr: ros1_bridge                                          
CMake Warning at CMakeLists.txt:151 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libyaml-cpp.so.0.6] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/opt/yaml_cpp_vendor/lib
    runtime library [liborocos-kdl.so.1.4] in /usr/lib may be hidden by files in:
      /opt/ros/foxy/lib
    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
[... 194 lines cut ...]
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

methylDragon, 2022-05-19:
Thank you so much for listing out the environment! Let me take a look

> Hi there! I'm having the same issue.... For now I have the same results. I'm using:
> 
> * Ubuntu 20.04
> * ROS1 Noetic (installed with apt)
> * ROS2 Foxy (installed with apt)
> * ROS Bridge installed with sources
> 
> I'll try something new, but I've been stuck with this same issue

methylDragon, 2022-05-19:
Hiya! I can't reproduce the issue, ros1_bridge completes its build for me on those environment configurations.

Could you:
- Ensure you followed:
```shell
$ colcon build --symlink-install --packages-skip ros1_bridge
$ source ${ROS1_INSTALL_PATH}/setup.bash
$ source ${ROS2_INSTALL_PATH}/setup.bash  # and any overlays
$ colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
```

- And also check if you've updated `python3-colcon-common-extensions`? I see a lot of allow-overriding warnings.
?

methylDragon, 2022-05-20:
Could I check if you started off with **JUST** ROS 2 sourced before running the first colcon build?

Can you delete your `build`, `install`, `log` directories and run a build?

JLBicho, 2022-05-23:
Hi again!
Thanks for the answer. I've tried that and these were the results:

```
$ source /opt/ros/foxy/setup.bash 
$ sudo rm -r build
$ sudo rm -r install/
$ sudo rm -r log/
$ colcon build --symlink-install --packages-skip ros1_bridge
                     
Summary: 0 packages finished [0.07s]
$ colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
[Processing: ros1_bridge]                            
[Processing: ros1_bridge]                                     
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                       
[Processing: ros1_bridge]                                         
[Processing: ros1_bridge]                                          
[Processing: ros1_bridge]                                          
--- stderr: ros1_bridge                                            
CMake Warning at CMakeLists.txt:145 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [liburdfdom_sensor.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu
    runtime library [liburdfdom_model_state.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu
    runtime library [liburdfdom_model.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu
    runtime library [liburdfdom_world.so.1.0] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /opt/ros/foxy/lib/x86_64-linux-gnu

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:191 (custom_executable)

CMake Warning at CMakeLists.txt:145 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/noetic/lib]
      dir 2 must precede it due to runtime library [liburdf.so]
    dir 1 is [/home/jose/ros_bridge/build/ros1_bridge]
    dir 2 is [/opt/ros/foxy/lib]
      dir 0 must precede it due to runtime library [libimage_transport.so]
    dir 3 is [/opt/ros/foxy/opt/yaml_cpp_vendor/lib]
    dir 4 is [/opt/ros/foxy/lib/x86_64-linux-gnu]
    dir 5 is [/opt/ros/foxy/opt/rviz_ogre_vendor/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
[... 104 lines cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/static_bridge.dir/build.make:440: static_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:233: CMakeFiles/static_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_1_to_2(controller_manager_msgs::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::msg::ControllerState_<std::allocator<void> >&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_2_to_1(controller_manager_msgs::msg::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::ControllerState_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/parameter_bridge.dir/build.make:440: parameter_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:314: CMakeFiles/parameter_bridge.dir/all] Error 2
/usr/bin/ld: warning: libconsole_bridge.so.0.4, needed by /opt/ros/noetic/lib/libcpp_common.so, may conflict with libconsole_bridge.so.1.0
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<controller_manager_msgs::ControllerState_<std::allocator<void> >, controller_manager_msgs::msg::ControllerState_<std::allocator<void> > >::convert_1_to_2(controller_manager_msgs::ControllerState_<std::allocator<void> > const&, controller_manager_msgs::msg::ControllerState_<std::allocator<void> >&)'
[... cut ...]
make[2]: *** [CMakeFiles/dynamic_bridge.dir/build.make:440: dynamic_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:341: CMakeFiles/dynamic_bridge.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ros1_bridge [11min 47s, exited with code 2]

Summary: 0 packages finished [11min 47s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

Results are the same. Could this be a conflict between libraries for ROS1 and ROS2? Maybe some apt update / upgrade? I've updated everything but the problem persists.

methylDragon, 2022-05-24:
Still no luck reproducing it on my machine or a docker file with your system specs.

Could you provide the the results of this please:
```
echo $LD_LIBRARY_PATH
```

And also, after cleaning your workspace again, rebuild with:
```
source /opt/ros/foxy/setup.bash 
sudo rm -r build install log
source /opt/ros/noetic/setup.bash 
source /opt/ros/foxy/setup.bash 
colcon build --symlink-install --packages-skip ros1_bridge
                     
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
```

methylDragon, 2022-05-25:
Alternatively, could you please provide a [minimal reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example) so I can verify and fix this bug? Otherwise we might be stuck here with me trying to debug the issue on your machine...

Something like a dockerfile with your setup, workspace sources, and commands to run that reproduce the problem inside the docker container could work.

methylDragon, 2022-05-26:
I thought it was `/usr/lib/x86_64-linux-gnu/gazebo-11/plugins`, but nope, it still builds on my end :(

```
Summary: 1 package finished [5min 25s]
methyldragon@methyldragon-MS-7885:~/Desktop/ros_workspaces/scratchpad_ws 
$ echo $LD_LIBRARY_PATH 
/opt/ros/noetic/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib
```

I'll just wait out for the minimal example, thanks for helping!

JLBicho, 2022-06-08:
Hi there!

So... I've tried to replicate my system in a Docker so I could send it as a minimal example... but in the Docker the compilation finished correctly :astonished: 
I think I might have something in my paths or some library not correctly updated.

In case someone wants the Dockerfile:

```
FROM ubuntu:20.04

RUN apt update

RUN locale && \
    apt update && \
    apt install locales -y && \
    locale-gen en_US en_US.UTF-8 && \ 
    update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 && \ 
    export LANG=en_US.UTF-8 && \ 
    locale

RUN apt update && apt install curl gnupg2 lsb-release -y 
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

RUN apt update && apt upgrade -y

RUN DEBIAN_FRONTEND=noninteractive TZ=Europe/Madrid apt-get -y install tzdata
RUN DEBIAN_FRONTEND=noninteractive apt-get -y install keyboard-configuration

RUN DEBIAN_FRONTEND=noninteractive apt install ros-foxy-desktop -y

RUN sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -

RUN apt update

RUN DEBIAN_FRONTEND=noninteractive apt install ros-noetic-desktop-full -y

RUN mkdir -p /home/rosbrige/src
WORKDIR /home/rosbrige
COPY ./build_rosbridge.bash /home/rosbrige/build_rosbridge.bash

RUN apt install -y git && apt install -y python3-colcon-common-extensions

RUN git clone https://github.com/ros2/ros1_bridge.git src

RUN bash build_rosbridge.bash

ENTRYPOINT [ "bash" ]

```

You will also need the script `build_rosbridge.bash`:

```
#!/bin/bash

# Source ros2 and build ws
source /opt/ros/foxy/setup.bash
colcon build --packages-skip ros1_bridge
# Source ros1
source /opt/ros/noetic/setup.bash
# Remove old folders
rm -r build
rm -r install
# Build bridge
colcon build --packages-select ros1_bridge --cmake-force-configure
# Source
source /home/rosbrige/install/local_setup.bash
# Check that it works
ros2 run ros1_bridge dynamic_bridge --print-pairs
```

I will keep on searching for what could be wrong in my system, but as I said I suspect it has something to do with a PATH badly written or something like that.
If someone has any idea of what it could be, I would kindly appreciate the help.

Also, thank you @methylDragon for your time and dedication!

methylDragon, 2022-06-08:
Alrighty! I'll close the issue for now, but if you happen to find out the reason for your issues, feel free to open it again, or post the solution here in case anyone else happens to have the same issue!

Cheers!

methylDragon, 2022-06-23:
> I have the same issue, but with Ubuntu 20.04, ROS 2 Galactic and ROS 1 Noetic. I can try to check if a clean ROS 1 and ROS 2 installation will help. Please write down what you are interested in to know! I will try to provide everything.

Heya!

As long as I can reproduce it on my end, we can make progress on this :o

Ideally a dockerfile that reproduces the issue would be great! Otherwise a description might not reproduce :>

ljaniec, 2022-06-24:
Okay, I don't know docker, but I can try to create it with the tutorials.

For now, I am trying one of the proposed solutions in other threads/places, with a fresh re-install of ROS1 and ROS2 + the same on the fresh system with Ubuntu 20.04

Eku127, 2025-01-10:
> CMake Warning

> Still no luck reproducing it on my machine or a docker file with your system specs.
> 
> Could you provide the the results of this please:
> 
> ```
> source /opt/ros/foxy/setup.bash 
> echo $LD_LIBRARY_PATH
> 
> source /opt/ros/noetic/setup.bash 
> source /opt/ros/foxy/setup.bash 
> echo $LD_LIBRARY_PATH
> ```
> 
> And also, after cleaning your workspace again, rebuild with `--event-handlers console_direct+`:
> 
> ```
> source /opt/ros/foxy/setup.bash 
> sudo rm -r build install log
> source /opt/ros/noetic/setup.bash 
> source /opt/ros/foxy/setup.bash 
> colcon build --symlink-install --packages-skip ros1_bridge
>                      
> colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --event-handlers console_direct+
> ```

Same problem but successully built with these commands

[image]