ros2/ros1_bridge #370: [question] YAML with mapping rules - which package should have it?

ljaniec, 2022-06-27:
Hello,

I am trying to bridge my custom ROS1 and ROS2 message packages, unfortunately, without much success.

My question:

- Where should [the mentioned `yaml` file](https://github.com/ros2/ros1_bridge/blob/galactic/doc/index.rst#how-can-i-specify-custom-mapping-rule-for-messages) be located? In the ROS1 message package or ROS2 message package or the `ros1_bridge` package? It is not clear to me in which one, because:

- By this part: https://github.com/ros2/ros1_bridge/blob/galactic/doc/index.rst#how-can-i-install-mapping-rule-files I think it should be `ros1_bridge` package and it's `package.xml` and `CMakeLists.txt`

- But by this example there: https://github.com/ros2/ros1_bridge/blob/galactic/doc/index.rst#how-can-i-install-mapping-rule-files the directory layout suggests it should be ROS2 message package?

Which version is correct? I think the documentation should be more precise there.

- Operating System:
Ubuntu 20.04, ROS 2 Galactic (binaries), ROS 1 Noetic (binaries), defualt DDS

quarkytale, 2022-06-27:
See [this](https://github.com/amifsud/fields_mapping) as an example, it was created for another issue but should help with file structure and installation. Note: `ros1_bridge` is a submodule in that repository, either `git submodule update` or manually clone it, if you're trying it out!

ljaniec, 2022-06-28:
Thank you for the answer. I hope some additional notes will be added to the documentation so it will be clear for everyone

quarkytale, 2022-06-28:
I will have that added, thanks!

closed 2022-06-28 by quarkytale