ros2/ros1_bridge #375: Param_Bridge cant parse --ros-args

geoeo, 2022-08-18:
```
Ros2 run ros1_bridge dynamic_bridge --ros-args -r __ns:=/test 
```

Works

```
Ros2 run ros1_bridge param_bridge --ros-args -r __ns:=/test 
```

Crashes. This is a big problem when trying to launch from a launchfile

gbiggs, 2022-08-18:
Please provide specific error output from the crash and as much detail about your environment as possible.

geoeo, 2022-08-19:
- System : Ubuntu 20.04
- ROS2: Galatic
- Installation Type: From Apt

The error after running `ros2 run ros1_bridge parameter_bridge --ros-args -r __ns:=/test ` 

```
[ERROR] [1660892133.744271987] [rcl]: Failed to parse global arguments
terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
  what():  failed to initialize rcl: Couldn't parse trailing -r flag. No remap rule found., at /tmp/binarydeb/ros-galactic-rcl-3.1.3/src/rcl/arguments.c:37
  ```
  
  I think this was already posted a year ago https://github.com/ros2/ros1_bridge/issues/316

clalancette, 2022-08-19:
Yes, agreed, that is the same error as in #316.  Closing this as a duplicate.

closed 2022-08-19 by clalancette