ros2/ros1_bridge #385: ros1_bridge fails to build

Anouar35, 2022-11-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - ros1_bridge from source. ROS1 Noetic and ROS2 Foxy

#### Steps to reproduce issue
1- Clone all workspaces from :  [](https://github.com/lFatality/ros2_bridge_custom_interfaces.git)
2- Clone [](https://github.com/ros2/ros1_bridge.git)to bridge_ws sources
3- Build ros1_ws and ros2_ws using respectively catkin_make and colcon build
4- After sourcing  
5 - Issue the following commands before building the ros1_bridge, by the same order:
`source /opt/ros/noetic/setup.bash
source /opt/ros/foxy/setup.bash 

# source the ROS workspaces
source <ros1_ws_dir>/devel/setup.bash
source <ros2_ws_dir>/install/setup.bash`

6 - Build the ros1_bridge using colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure

#### Expected behavior
It is expected that the ros1_bridge builds successfully. 

#### Actual behavior
Output from the console: 
`Starting >>> ros1_bridge
--- stderr: ros1_bridge                              
In file included from /home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/build/ros1_bridge/generated/custom_msg_ros2_factories.hpp:6,
                 from /home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/build/ros1_bridge/generated/custom_msg_ros2__msg__CustomMessage__factories.cpp:3:
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:30: error: ‘SharedLibrary’ is not a member of ‘rcpputils’
  358 |   std::shared_ptr<rcpputils::SharedLibrary> ts_lib_;
      |                              ^~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:30: error: ‘SharedLibrary’ is not a member of ‘rcpputils’
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:43: error: template argument 1 is invalid
  358 |   std::shared_ptr<rcpputils::SharedLibrary> ts_lib_;
      |                                           ^
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp: In constructor ‘ros1_bridge::Factory<ROS1_T, ROS2_T>::Factory(const string&, const string&)’:
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:46:23: error: ‘get_typesupport_library’ is not a member of ‘rclcpp’
   46 |     ts_lib_ = rclcpp::get_typesupport_library(ros2_type_name, "rosidl_typesupport_cpp");
      |                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:48:31: error: ‘get_typesupport_handle’ is not a member of ‘rclcpp’
   48 |       type_support_ = rclcpp::get_typesupport_handle(
      |                               ^~~~~~~~~~~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:50:10: error: invalid type argument of unary ‘*’ (have ‘int’)
   50 |         *ts_lib_);
      |          ^~~~~~~
In file included from /home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/build/ros1_bridge/generated/action_msgs_factories.hpp:6,
                 from /home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/build/ros1_bridge/generated/action_msgs_factories.cpp:3:
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:30: error: ‘SharedLibrary’ is not a member of ‘rcpputils’
  358 |   std::shared_ptr<rcpputils::SharedLibrary> ts_lib_;
      |                              ^~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:30: error: ‘SharedLibrary’ is not a member of ‘rcpputils’
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:43: error: template argument 1 is invalid
  358 |   std::shared_ptr<rcpputils::SharedLibrary> ts_lib_;
      |                                           ^
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp: In constructor ‘ros1_bridge::Factory<ROS1_T, ROS2_T>::Factory(const string&, const string&)’:
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:46:23: error: ‘get_typesupport_library’ is not a member of ‘rclcpp’
   46 |     ts_lib_ = rclcpp::get_typesupport_library(ros2_type_name, "rosidl_typesupport_cpp");
      |                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:48:31: error: ‘get_typesupport_handle’ is not a member of ‘rclcpp’
   48 |       type_support_ = rclcpp::get_typesupport_handle(
      |                               ^~~~~~~~~~~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:50:10: error: invalid type argument of unary ‘*’ (have ‘int’)
   50 |         *ts_lib_);
      |          ^~~~~~~
In file included from /home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/builtin_interfaces_factories.hpp:29,
[... 105 lines cut ...]
      |                              ^~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:30: error: ‘SharedLibrary’ is not a member of ‘rcpputils’
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:358:43: error: template argument 1 is invalid
  358 |   std::shared_ptr<rcpputils::SharedLibrary> ts_lib_;
      |                                           ^
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp: In constructor ‘ros1_bridge::Factory<ROS1_T, ROS2_T>::Factory(const string&, const string&)’:
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:46:23: error: ‘get_typesupport_library’ is not a member of ‘rclcpp’
   46 |     ts_lib_ = rclcpp::get_typesupport_library(ros2_type_name, "rosidl_typesupport_cpp");
      |                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:48:31: error: ‘get_typesupport_handle’ is not a member of ‘rclcpp’
   48 |       type_support_ = rclcpp::get_typesupport_handle(
      |                               ^~~~~~~~~~~~~~~~~~~~~~
/home/ac272844/ros2_bridge_custom_interfaces/bridge_ws/src/ros1_bridge/include/ros1_bridge/factory.hpp:50:10: error: invalid type argument of unary ‘*’ (have ‘int’)
   50 |         *ts_lib_);
      |          ^~~~~~~
make[2]: *** [CMakeFiles/ros1_bridge.dir/build.make:1216 : CMakeFiles/ros1_bridge.dir/generated/action_msgs__msg__GoalStatusArray__factories.cpp.o] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:227 : CMakeFiles/ros1_bridge.dir/all] Erreur 2
make: *** [Makefile:146 : all] Erreur 2
---
Failed   <<< ros1_bridge [15.1s, exited with code 2]

Summary: 0 packages finished [15.3s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
`

gbiggs, 2022-11-10:
The `ros1_bridge` will only consider interface definitions from packages with names ending in `_msg` or `_interfaces`. You need to fix the package names for your packages.

gbiggs, 2022-11-13:
If you want to build `ros1_bridge` with Foxy then you need to use the [Foxy branch](https://github.com/ros2/ros1_bridge/tree/foxy). The `master` branch only works with Rolling.

gabriel-azeredo, 2022-11-17:
Enter the command `git checkout foxy` after cloning the repo to be sure that you are building the correct version

clalancette, 2022-12-01:
Given the above, I'm going to close this out.  If you are still having issues, please feel free to reopen.

closed 2022-12-01 by clalancette

tugbakara, 2023-05-29:
> If you want to build `ros1_bridge` with Foxy then you need to use the [Foxy branch](https://github.com/ros2/ros1_bridge/tree/foxy). The `master` branch only works with Rolling.

Is it still valid?

rmssalgueiro, 2024-02-26:
> zhuhj
Where can I find the commit 81f8b? The commit numbers are usually bigger, with 7 digits

Tim-HW, 2025-03-06:
Update ! I had the same error as you. The issues seems to come from the fact that I sourced **foxy** and **noetic** before performing the **colcon build**. **I reopened another terminal, only sourced foxy and everything worked !**

Cheers