ros2/ros1_bridge #387: Package 'ros1_bridge` not found

jacobsayono, 2022-11-30:
Hello,

I was successful in building ros1_bridge with colcon (took about 20 mins). I am trying to do the first example. Note that I am bridging neotic (ros1) to foxy (ros2) so I have cloned the ros1_bridge repository from the foxy tree.

In shell A, I sourced ros1 then ran roscore. In shell B, I sourced ros1 then sourced ros2. Then when I try to run: `ros2 run ros1_bridge dynamic_bridge`, i get the error `Package 'ros1_bridge' not found`.

Am I missing anything here?

Thanks!

gbiggs, 2022-12-01:
Thanks for your question. However we ask that you please ask questions on http://answers.ros.org following our support guidelines: http://wiki.ros.org/Support .

closed 2022-12-01 by gbiggs

jacobsayono, 2022-12-01:
Thanks, I have asked in ROS support. However, this is an issue with ros1_bridge, not ROS (my ros1 and ros2 works perfectly); thought some folks here would have a solution to this issue.

DanBmh, 2023-04-19:
I had the same problem and could solve it with the solution @jacobsayono posted in the _ROS_ forum (https://answers.ros.org/question/410063/package-ros1_bridge-not-found/), even though I had to slightly adjust the path.

In my case it was `source /project/dev_ws/src/install/ros1_bridge/share/ros1_bridge/local_setup.bash` with `/project/dev_ws/` being my overlay workspace in which I have cloned the _ros1_bridge_ repository (found the file by running `find /project/dev_ws/ -name "*.bash"` after building the _bridge_).