ros2/ros1_bridge #390: Build Error about message files

iemirosman, 2022-12-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 
Ros 1 : noetic 
Ros 2 : humble 

#### Steps to reproduce issue
I followed the build instructions and it gives fatal error: 
```
admin@ubuntu:/workspaces/ros1_ws/ros1_bridge$ colcon build --packages-select ros1_bridge --cmake-force-configure
Starting >>> ros1_bridge
[Processing: ros1_bridge]                         
[Processing: ros1_bridge]                                    
[Processing: ros1_bridge]                                      
[Processing: ros1_bridge]                                      
[Processing: ros1_bridge]                                      
[Processing: ros1_bridge]                                      
--- stderr: ros1_bridge                                      
CMake Warning at CMakeLists.txt:207 (add_library):
  Cannot generate a safe runtime search path for target ros1_bridge because
  there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/humble/install/lib]
      dir 1 must precede it due to runtime library [libimage_transport.so]
    dir 1 is [/opt/ros/noetic/lib]
      dir 0 must precede it due to runtime library [libtf2_ros.so]
    dir 2 is [/opt/ros/humble/install/opt/rviz_ogre_vendor/lib]

  Some of these libraries may not be found correctly.

CMake Warning at /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:50 (add_executable):
  Cannot generate a safe runtime search path for target test_convert_generic
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/noetic/lib]
      dir 2 must precede it due to runtime library [libtf2_ros.so]
    dir 1 is [/workspaces/ros1_ws/ros1_bridge/build/ros1_bridge]
    dir 2 is [/opt/ros/humble/install/lib]
      dir 0 must precede it due to runtime library [libimage_transport.so]
    dir 3 is [/opt/ros/humble/install/opt/rviz_ogre_vendor/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest_executable.cmake:37 (_ament_add_gtest_executable)
  /opt/ros/humble/install/share/ament_cmake_gtest/cmake/ament_add_gtest.cmake:68 (ament_add_gtest_executable)
  CMakeLists.txt:92 (ament_add_gtest)

CMake Warning at CMakeLists.txt:178 (add_executable):
  Cannot generate a safe runtime search path for target static_bridge because
  there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/noetic/lib]
      dir 2 must precede it due to runtime library [libtf2_ros.so]
    dir 1 is [/workspaces/ros1_ws/ros1_bridge/build/ros1_bridge]
    dir 2 is [/opt/ros/humble/install/lib]
      dir 0 must precede it due to runtime library [libimage_transport.so]
    dir 3 is [/opt/ros/humble/install/opt/rviz_ogre_vendor/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:224 (custom_executable)

CMake Warning at CMakeLists.txt:178 (add_executable):
  Cannot generate a safe runtime search path for target parameter_bridge
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/noetic/lib]
      dir 2 must precede it due to runtime library [libtf2_ros.so]
    dir 1 is [/workspaces/ros1_ws/ros1_bridge/build/ros1_bridge]
    dir 2 is [/opt/ros/humble/install/lib]
      dir 0 must precede it due to runtime library [libimage_transport.so]
    dir 3 is [/opt/ros/humble/install/opt/rviz_ogre_vendor/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:231 (custom_executable)

CMake Warning at CMakeLists.txt:178 (add_executable):
  Cannot generate a safe runtime search path for target dynamic_bridge
  because there is a cycle in the constraint graph:

    dir 0 is [/opt/ros/noetic/lib]
      dir 2 must precede it due to runtime library [libtf2_ros.so]
    dir 1 is [/workspaces/ros1_ws/ros1_bridge/build/ros1_bridge]
    dir 2 is [/opt/ros/humble/install/lib]
      dir 0 must precede it due to runtime library [libimage_transport.so]
    dir 3 is [/opt/ros/humble/install/opt/rviz_ogre_vendor/lib]

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:238 (custom_executable)

In file included from /workspaces/ros1_ws/ros1_bridge/build/ros1_bridge/generated/get_factory.cpp:42:
/workspaces/ros1_ws/ros1_bridge/build/ros1_bridge/generated/visualization_msgs_factories.hpp:18:10: fatal error: visualization_msgs/MeshFile.h: No such file or directory
   18 | #include <visualization_msgs/MeshFile.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ros1_bridge.dir/build.make:1499: CMakeFiles/ros1_bridge.dir/generated/get_factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:338: CMakeFiles/ros1_bridge.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< ros1_bridge [3min 8s, exited with code 2]

Summary: 0 packages finished [3min 9s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge

```
Help plz.

yijh0611, 2023-04-28:
Same error

harshkakashaniya, 2023-05-30:
Any Update ...
Facing same error

hsurya08, 2023-06-08:
Getting the same issue. Any help would be appreciated. Thanks!

Wilson-shuo, 2023-08-14:
Hi developer! I meet the same problem when I was building a ros1_bridge on a nvidia-docker which is based on ros2 humble.

I found that problem is caused by your third-party ros2 package which customized some topics that ros do not support.(Meshfile .etc) 

The most "not that smart" solution is remove all third-party ros2 package and build bridge again.

I still hope they can make a easy-to-use blacklist to avoid ros1-bridge select all ros2 package as dependent.

JTaveau, 2023-11-05:
Any news on this issue? I'm facing the same problem.

ChrisBlitz832, 2024-01-04:
I'm facing the same error. I am using a Jetson Orin with the Isaac Ros Buildfarm.

rvxfahim, 2024-04-24:
same issue