ros2/ros1_bridge #393: [Humble] Building Failed

Ogunniran, 2023-01-18:
## Bug report

**Note:** It built properly after c2fa9ed11d1be6cb36441ad317d0797deb4714d9 was removed.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
  - ROS1: Noetic
  - ROS2: Humble
- Installation type:
  - From source

Building the ros1_bridge with `colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure` failed with the error:

```
//home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp: In instantiation of ‘void ros1_bridge::streamPrimitiveVector(ros::serialization::OStream&, const VEC_PRIMITIVE_T&) [with VEC_PRIMITIVE_T = std::vector<bool, std::allocator<bool> >]’:
/home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp:830:24:   required from here
/home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp:694:46: error: lvalue required as unary ‘&’ operand
  694 |   memcpy(stream.advance(data_len), &vec.front(), data_len);
      |                                     ~~~~~~~~~^~
/home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp: In instantiation of ‘void ros1_bridge::streamPrimitiveVector(ros::serialization::IStream&, VEC_PRIMITIVE_T&) [with VEC_PRIMITIVE_T = std::vector<bool, std::allocator<bool> >]’:
/home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp:1063:24:   required from here
/home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp:711:20: error: taking address of rvalue [-fpermissive]
  711 |   memcpy(&vec.front(), stream.advance(data_len), data_len);
      |           ~~~~~~~~~^~
/home/x/ros1_bridge/ros_bridge/build/ros1_bridge/generated/...__factories.cpp:711:9: warning: ‘void* memcpy(void*, const void*, size_t)’ writing to an object of type ‘std::vector<bool, std::allocator<bool> >::reference’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  711 |   memcpy(&vec.front(), stream.advance(data_len), data_len);
```

#### Additional information

----

One of the ROS2 msgs which triggered the error has a similar format:

```
std_msgs/Header header
# foo
string[] foo_1
float64[] foo_2
float64[] foo_3
float64[] foo_4
int32[] foo_5
float64[] foo_6
int32[] foo_7
bool[] foo_8
bool[] foo_9

int32[] foo_x
int32[] foo_y
```

clalancette, 2023-02-02:
How are you using Noetic on Ubuntu 22.04?  Did you build from source.

While this should probably work, this isn't currently a supported configuration, so we may not be able to help here.

Ogunniran, 2023-02-03:
Ok, thanks for the reply. Yes it was built from source.

Ogunniran, 2023-02-17:
The issue still occurs while building with rolling and noetic, using the [docker image](https://hub.docker.com/layers/library/ros/rolling-ros1-bridge-focal/images/sha256-0998977900af0fb9c48136668385e05f25e82a3bfa06f69e83de5af60c477700?context=explore).

alexg-k, 2023-03-16:
Hi, I have the same setup and the same error message as described above. Just want to add that this error seemed to be introduced with the latest commits. When I checkout the commit ID `b9f1739fd84fc877a8ec6e5c416b65aa2d782f89` for example and build the otherwise untouched project again, it runs fine without the error from above.

Timple, 2023-03-16:
This is great!

This commit also 'fixes' #391

oKermorgant, 2023-07-06:
Hi,

I have just gone into the same issue.
My guess is that `vector<bool>` is a bad class and is not a vector of bool.

As a consequence, taking the address of the first element does not work as it should, because there are no bools inside a `vector<bool>`. Returning by value is fine but a reference is not possible.

I have written a quick fix, need to test it before submitting a PR.