ros2/ros1_bridge #394: No template specialization for the pair

mertbacaksiz, 2023-01-29:
I use neotic as Ros1 and galactic as Ros2. I created a publisher from the example_interface package. And I want to subscribe it with Ros1 std_msgs package. But the error I get is as follows: with ROS 2 type 'example_interfaces/msg/Float64' and ROS 1 type '': No template specialization for the pair. Can you help me ?

[image]

Ogunniran, 2023-02-03:
Seems it was not built properly, or maybe your workspace was not properly source, maybe you could rebuild based on the README instruction and source your bridge ws before running the script.

mertbacaksiz, 2023-02-03:
> Seems it was not built properly, or maybe your workspace was not properly source, maybe you could rebuild based on the README instruction and source your bridge ws before running the script.

I solved it by directly installing noetic own package.
Thank you.

closed 2023-02-03 by mertbacaksiz