ros2/ros1_bridge #403: follow joint trajectory action

MattiaDeiRossi, 2023-05-10:
Hi all, 

is there anyone that has tried to use a server action on ros2 to communicate with client action on ros1?

I'm trying to use ros2_control framework with follow joint trajectory action server. I would connect to the same action with move_group client on ros1.

Have you got any ideas?
Any suggestions will be appreciated.
Thanks.

shobhitagg, 2023-08-25:
Same issue here. Did you find any solution?

MattiaDeiRossi, 2023-08-27:
I didn't find any quick solution, so I decided to port all my code in ROS2 instead using ros1_bridge.

gavanderhoorn, 2023-08-27:
#256?

shobhitagg, 2023-08-27:
> 

I am leaning towards that too

shobhitagg, 2023-08-27:
> #256?

I may not have understood the conversation there (It has a lot going on in there). But it seemed as if the effort was implemented with `dynamic_bridge`. Is there any documentation, sample, or even a draft set of instructions that we can use as a starting point? 
Happy to help with documentation, with motoman_gp4 example, if I can make it work.

shobhitagg, 2023-08-28:
> #256?

I assumed it was merged with the main repo. But apparently it is not.

I tried cloning the fork repo from here [https://github.com/ipa-hsd/ros1_bridge.git](https://github.com/ipa-hsd/ros1_bridge.git).
The package doesn't compile successfully. I am using ROS Noetic and ROS2 Humble. Following was the error message.

```shell
--- stderr: ros1_bridge
In file included from /home/mfi/repos/ros2_ws/src/ros1_bridge/src/dynamic_bridge.cpp:38:
/home/mfi/ros2_humble/install/rclcpp/include/rclcpp/rclcpp/scope_exit.hpp:24:2: warning: #warning rclcpp/scope_exit.hpp has been deprecated, please use rcpputils/scope_exit.hpp instead [-Wcpp]
   24 | #warning rclcpp/scope_exit.hpp has been deprecated, please use rcpputils/scope_exit.hpp instead      |  ^~~~~~~
In file included from /home/mfi/repos/ros2_ws/src/ros1_bridge/src/dynamic_bridge.cpp:40:
/home/mfi/ros2_humble/src/ros2/rcutils/include/rcutils/get_env.h:25:2: warning: #warning rcutils/get_env.h has been deprecated, please include rcutils/env.h instead [-Wcpp]
   25 | #warning rcutils/get_env.h has been deprecated, please include rcutils/env.h instead
      |  ^~~~~~~
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_rosbag2_storage_mcap_testdata__msg__ComplexIdl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_rosbag2_storage_mcap_testdata__msg__BasicIdl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/parameter_bridge.dir/build.make:507: parameter_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:342: CMakeFiles/parameter_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_rosbag2_storage_mcap_testdata__msg__ComplexIdl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_rosbag2_storage_mcap_testdata__msg__BasicIdl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/static_bridge.dir/build.make:507: static_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:316: CMakeFiles/static_bridge.dir/all] Error 2
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_rosbag2_storage_mcap_testdata__msg__ComplexIdl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: libros1_bridge.so: undefined reference to `ros1_bridge::get_factory_rosbag2_storage_mcap_testdata__msg__BasicIdl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dynamic_bridge.dir/build.make:507: dynamic_bridge] Error 1
make[1]: *** [CMakeFiles/Makefile2:368: CMakeFiles/dynamic_bridge.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< ros1_bridge [2min 49s, exited with code 2]

Summary: 0 packages finished [2min 49s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge
```

shobhitagg, 2023-08-28:
If I pull the changes made in the PR 256, I still am not able to compile. 
[error.txt](https://github.com/ros2/ros1_bridge/files/12450094/error.txt)

smith-doug, 2023-12-18:
> If I pull the changes made in the PR 256, I still am not able to compile. [error.txt](https://github.com/ros2/ros1_bridge/files/12450094/error.txt)

The signature of goal_response_callback changed in humble.  I'm working on a PR and still need to do a proper test.  Something like the following worked in a brief test with noetic/humble on 22.04, with noetic in a conda env.

```
send_goal_ops.goal_response_callback =
        [this, &gh2_future](const ROS2GoalHandle & gh2) mutable {
          auto & goal_handle = gh2;
          if (!goal_handle) {
            gh1_.setRejected();          // goal was not accepted by remote server
            return;
          }

          gh1_.setAccepted();

          {
            std::lock_guard<std::mutex> lock(mutex_gh_);
            gh2_ = goal_handle;

            if (canceled_) {          // cancel was called in between
              auto fut = client_->async_cancel_goal(gh2_);
            }
          }
        };
```

smith-doug, 2024-01-31:
I've gotten the action bridge working well enough with Noetic/Humble to execute motions on a motoman HC10.  (I know there's a ros2 motoman driver, but it is currently set up for ros1).  It's still a wip and I wouldn't trust it for anything serious.

https://github.com/smith-doug/ros1_bridge/tree/action_bridge_humble  
Currently requires https://github.com/smith-doug/bridge_mapping as I didn't want to hardcode any mappings, but that might not be possible.  Message types like Log change both the number of fields, and the values of the defined log severity levels.