ros2/ros1_bridge #404: libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >

kenloumixx, 2023-05-16:
## Bug report

**Required Info:**

- Operating System: ubuntu 18.04, ros1 noetic, ros2 galactic
  -  
- Installation type: source
  -  
- Version or commit hash: 
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
source /opt/ros/noetic/setup.zsh
source /opt/ros/galactic/setup.zsh
```

#### Expected behavior
clean build

#### Actual behavior
```
Starting >>> ros1_bridge
[Processing: ros1_bridge]                              
[Processing: ros1_bridge]                                      
--- stderr: ros1_bridge                                          
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection2D_<std::allocator<void> > const&, vision_msgs::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection2D_<std::allocator<void> > const&, vision_msgs::msg::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection3D_<std::allocator<void> > const&, vision_msgs::Detection3D_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [dynamic_bridge] Error 1
make[1]: *** [CMakeFiles/dynamic_bridge.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection2D_<std::allocator<void> > const&, vision_msgs::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection2D_<std::allocator<void> > const&, vision_msgs::msg::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection3D_<std::allocator<void> > const&, vision_msgs::Detection3D_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [parameter_bridge] Error 1
make[1]: *** [CMakeFiles/parameter_bridge.dir/all] Error 2
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection3D_<std::allocator<void> > const&, vision_msgs::msg::Detection3D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection2D_<std::allocator<void> > const&, vision_msgs::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection2D_<std::allocator<void> >, vision_msgs::msg::Detection2D_<std::allocator<void> > >::convert_1_to_2(vision_msgs::Detection2D_<std::allocator<void> > const&, vision_msgs::msg::Detection2D_<std::allocator<void> >&)'
libros1_bridge.so: undefined reference to `ros1_bridge::Factory<vision_msgs::Detection3D_<std::allocator<void> >, vision_msgs::msg::Detection3D_<std::allocator<void> > >::convert_2_to_1(vision_msgs::msg::Detection3D_<std::allocator<void> > const&, vision_msgs::Detection3D_<std::allocator<void> >&)'
collect2: error: ld returned 1 exit status
make[2]: *** [static_bridge] Error 1
make[1]: *** [CMakeFiles/static_bridge.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< ros1_bridge [1min 19s, exited with code 2]

Summary: 0 packages finished [1min 20s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge

```
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SuneSoerensen, 2023-06-08:
I have the same problem, only I'm using Ubuntu 20.04. I am able to compile the bridge without having ```vision_msgs``` installed (both the noetic and galactic versions). Is that also the case for you?

kenloumixx, 2023-06-08:
@SuneSoerensen Yes, in my case the detection related messages in `vision_msgs` matter, so I deleted manually Detection2D, Detection3D, Detection2DArray, Detection3DArray in ros galactic, though I don't recommend this way

Hugal31, 2024-07-23:
I have the same issue with `controller_manager_msgs::ControllerState`. It seems when the message are not compatible, but required by a srv definition, the srv definition references the factory for the incompatible message.
It would be nice to be able to allowlist/denylist the messages or packages.