ros2/ros1_bridge #420: Unstable behavior of   std::bad_alloc

AssemSadek, 2023-09-27:
## Bug report

**Required Info:**

- Operating System:  Ubuntu 20.04 jetpack 5.1 on Jetson Xavier
- Installation type: docker image on arm64v8/ros:foxy-ros1-bridge-focal
- DDS implementation: Fast-RTPS

#### Steps to reproduce issue
1. docker run -it --rm --net host arm64v8/ros:foxy-ros1-bridge-focal bash 
2. source ros_entrypoint.sh
3. ros2 run ros1_bridge dynamic_bridge --bridge-all-topics

#### Expected behavior

#### Actual behavior
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

#### Additional Information

This behavior is not reproduced at each time. But it happens frequently at unexpected moments. But I noticed some triggers:
1) at the moment of launching the node
2) at the moment of a new subscriber to a topic. The same topic can have already other subscribers or zero subscribers before this new subscription. Therefore, the number of subscribers is not an indicator.
3) whether I use the flag --bridge-all-topics or not, the behavior can pop out in both scenarios. Therefore, the number of bridges done is not also an indicator.

F1iX, 2023-12-18:
For us, the `std::bad_alloc` problem appeared when using a ROS2 *Foxy* bridge to transfer `sensor_msgs/NavSatFix` messages from a ROS2 *Iron* container. Looks like the ROS2 versions are not necessarily compatible.

gavanderhoorn, 2023-12-18:
Iron is not compatible with anything else IIRC. Mostly/largely because of https://github.com/ros2/rmw_dds_common/pull/68.