ros2/ros1_bridge #422: ros1_bridge dashing problem

iqraaslam01, 2023-10-23:
I'm trying to build ros1_bridge for couldn't build it. instead getting the following error: 
[image]
Does any one of you have faced the same issue?

fujitatomoya, 2023-10-23:
@gitgeek29 Unfortunately Dashing is already E.O.L(May 2021), could you try supported distribution instead?

see more details for https://docs.ros.org/en/rolling/Releases.html.

sloretz, 2023-11-02:
Closing as Dashing is EOL.

closed 2023-11-02 by sloretz

dsyddn, 2024-03-29:
static_assert is a new macro in C++11.
try to modify 'static_assert' to 'assert' included in C header <assert.h>.
(tips: modify the .em file in resource folder , which generate cpp file during buiding)