ros2/ros1_bridge #425: ROS2 Foxy: bridge causing continuous  errors from realsense on depth image

Luke1962, 2023-11-28:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04 , HW Jetson Nano 4GB
- Installation type: 
- ros1-bridge: from atp ( ros-foxy-ros1-bridge/focal,now 0.9.7-1focal.20230527.083653 arm64 [installed])
- realsense2_camera:  from source ()
- Librealsense: from sources

#### Steps to reproduce issue
run ros1_bridge with `source /opt/ros/foxy/setup.bash && ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`
run realsense2_camera built with ros2 launch realsense2_camera rs_launch.py, where parameters in launch file are the following (depth z16 + image_color_raw):
```
configurable_parameters = [{'name': 'camera_name',                  'default': 'realsense', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "''", 'description': 'yaml config file'},
                           {'name': 'enable_pointcloud',            'default': 'false', 'description': 'enable pointcloud'},
                           {'name': 'unite_imu_method',             'default': "''", 'description': '[copy|linear_interpolation]'},
                           {'name': 'json_file_path',               'default': "''", 'description': 'allows advanced configuration'},
                           {'name': 'log_level',                    'default': 'info', 'description': 'debug log level [DEBUG|INFO|WARN|ERROR|FATAL]'},
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},
                         #   {'name': 'depth_width',                  'default': '-1', 'description': 'depth image width'},
                        #    {'name': 'depth_height',                 'default': '-1', 'description': 'depth image height'},
                           {'name': 'depth_width',                  'default': '424', 'description': 'depth image width'},
                           {'name': 'depth_height',                 'default': '240', 'description': 'depth image height'},
                           {'name': 'publish_tf',                 'default': 'false', 'description': 'enable tf publish'},
                          {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                        #    {'name': 'color_width',                  'default': '-1', 'description': 'color image width'},
                        #    {'name': 'color_height',                 'default': '-1', 'description': 'color image height'},
                           {'name': 'color_width',                  'default': '640', 'description': 'color image width'},
                           {'name': 'color_height',                 'default': '480', 'description': 'color image height'},
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'infra_width',                  'default': '-1', 'description': 'infra width'},
                           {'name': 'infra_height',                 'default': '-1', 'description': 'infra width'},
                           {'name': 'enable_infra1',                'default': 'true', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'true', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'fisheye_width',                'default': '-1', 'description': 'fisheye width'},
                           {'name': 'fisheye_height',               'default': '-1', 'description': 'fisheye width'},
                           #{'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye1',              'default': 'false', 'description': 'enable fisheye1 stream'},
                           #{'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'enable_fisheye2',              'default': 'false', 'description': 'enable fisheye2 stream'},
                           {'name': 'confidence_width',             'default': '-1', 'description': 'depth image width'},
                           {'name': 'confidence_height',            'default': '-1', 'description': 'depth image height'},
                        #   {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'enable_confidence',            'default': 'false', 'description': 'enable depth stream'},
                           {'name': 'fisheye_fps',                  'default': '-1.', 'description': ''},
                        #    {'name': 'depth_fps',                    'default': '-1.', 'description': ''},
                           {'name': 'depth_fps',                    'default': '15.', 'description': ''},
                           {'name': 'confidence_fps',               'default': '-1.', 'description': ''},
                           {'name': 'infra_fps',                    'default': '-1.', 'description': ''},
                        #    {'name': 'color_fps',                    'default': '-1.', 'description': ''},
                           {'name': 'color_fps',                    'default': '15.', 'description': ''},
                           {'name': 'gyro_fps',                     'default': '-1.', 'description': ''},
                           {'name': 'accel_fps',                    'default': '-1.', 'description': ''},
                           {'name': 'color_qos',                    'default': 'SYSTEM_DEFAULT', 'description': 'QoS profile name'},
                           {'name': 'confidence_qos',               'default': 'SYSTEM_DEFAULT', 'description': 'QoS profile name'},
                           {'name': 'depth_qos',                    'default': 'SYSTEM_DEFAULT', 'description': 'QoS profile name'},
                           {'name': 'fisheye_qos',                  'default': 'SYSTEM_DEFAULT', 'description': 'QoS profile name'},
                           {'name': 'infra_qos',                    'default': 'SYSTEM_DEFAULT', 'description': 'QoS profile name'},
                           {'name': 'pointcloud_qos',               'default': 'SYSTEM_DEFAULT', 'description': 'QoS profile name'},
                           {'name': 'enable_gyro',                  'default': 'false', 'description': ''},
                           {'name': 'enable_accel',                 'default': 'false', 'description': ''},
                           {'name': 'enable_pose',                  'default': 'false', 'description': ''},
                           {'name': 'pointcloud_texture_stream',    'default': 'RS2_STREAM_COLOR', 'description': 'testure stream for pointcloud'},
                           {'name': 'pointcloud_texture_index',     'default': '0', 'description': 'testure stream index for pointcloud'},
[... 10 lines cut ...]
                           {'name': 'topic_odom_in',                'default': "''", 'description': 'topic for T265 wheel odometry'},
                           {'name': 'tf_publish_rate',              'default': '0.0', 'description': 'Rate of publishing static_tf'},
                           {'name': 'diagnostics_period',           'default': '0.0', 'description': 'Rate of publishing diagnostics. 0=Disabled'},
                           {'name': 'rosbag_filename',              'default': "''", 'description': 'A realsense bagfile to run from as a device'},
                           {'name': 'temporal.holes_fill',          'default': '0', 'description': 'Persistency mode'},
                           {'name': 'stereo_module.exposure.1',     'default': '7500', 'description': 'Initial value for hdr_merge filter'},
[... cut ...] b8).**
If I kill bridge node , immediately the above error messages are not anymore produced by realsense node.
#### Additional information
No depth subscribers on ROS1 side
May be I'm doing something wrong , but my understanding is that ros1_bridge is subscribing depth as rgb8 instead of Z16 image format. 
My question is : if it is not a bug, is there a way to change this behavior without removing the " --bridge-all-topics" option ? (since I still have a lot of  ROS1 messages not yet migrates).

yijh0611, 2024-02-06:
Simillar error.

paulijosey, 2024-02-09:
same issue here!

christophebedard, 2024-02-09:
ROS 2 Foxy is not supported anymore. You should try with a supported distro: https://docs.ros.org/en/rolling/Releases.html#list-of-distributions