ros2/ros1_bridge #428: ros bridge is crashing as soon as I try to run it.

manishsaini0901, 2024-02-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

1. Source the noetic environment
2. Source the foxy environment
3. launch the roscore in another terminal (after sourcing noetic)
4. run the command: "ros2 run ros1_bridge dynamic_bridge"

#### Expected behavior

Should start bridging the topics

#### Actual behavior

Simply crash and don't give any reason for the crash or anything that is displayed.

mjcarroll, 2024-02-20:
This needs more information in order to be properly diagnosed.  Can you run it under a debugger to try to get more information about how the crash is happening?

OskaOsinski, 2024-04-19:
Hi i have similar issue. 
**Required Info:**
- Operating System:
    - Ubuntu 20.04
- Installation type:
    - binaries
- Version or commit hash:
    - x
- DDS implementation:
    - Fast-RTPS
- Client library (if applicable):
    - N/A

**Steps to reproduce issue**
Source the noetic environment
Source the humble environment
launch the roscore in another terminal (after sourcing noetic)
run the command: "ros2 run ros1_bridge dynamic_bridge"

**Expected behavior**
Should start and wait in the loop for topics to be bridged

**Actual behavior**
Application closes as fast as the command is executed

After a while trying to figuring out why it closes without any output, even an error would be appreciated. So, I decided to go into the build directory of my workspace where the ros_1 bridge was built. I found the binary form of dynamic_bridge and after executing it, it works perfectly fine. So my problem is probably somewhere in the linking process, I guess.