ros2/ros1_bridge #436: Build with ROS2 Jazzy and ROS noetic

Hunk86, 2024-06-26:
**Required Info:**

- Operating System:
  - Ubuntu 20.04 but also docker with ubuntu 24.04
  - ROS noetic and ROS2 Jazzy

I want to build the ROS1 bridge by myself and I've read the documentation. But for the most important part I didn't find anything on the documentation, it only mentioned that you should define your ROS1 and ROS2 install space

Now my question: ROS 1 noetic is official supported till Ubuntu 20.04 and ROS2 jazzy with Ubuntu 24.04

How i could install both?

I tried two different solution for me

**1. Setup everything in Ubuntu 20.04 with binary install of ros2**

At the moment I tried to build it with a ubuntu 20.04 installed ROS noetic and with the binary Release of ROS2

But when i build it I got linking problem because of traecetools which I can't disable

I tried :

```
colcon build --symlink-install --packages-select ros1_bridge --cmake-force-configure --cmake-args -DBUILD_TESTING=OFF -DTRACETOOLS_DISABLED=ON 
```

I always get

Package 'tracetools' exports library 'lttng-ust-common' which couldn't be
```
/usr/bin/ld: cannot find -llttng-ust-common
/usr/bin/ld/usr/bin/ld: cannot find : cannot find -llttng-ust-common-llttng-ust-common
```

2. Build inside docker container

I'm using docker run -it osrf/ros:jazzy-desktop

I just copied /opt/noetic from my host system

But still then I've problems with e.g. the correct boost version for noetic 

Are my thoughts correct to build it like that ? I dont understand why I can't disable the tracetool, or what is the best way to build ros bridge by yourself?

jtaveau, 2024-07-17:
Hey @Hunk86, did you manage to solve the second issue?

jaehho, 2024-10-23:
im having the same issue, and I think the problem is that the bridge needs to run on a system with both ros1 noetic and ros2 jazzy, but there is no ubuntu distribution that supports both

BlueNoise, 2024-11-07:
I am also interested in this. I am trying to port ROS1 noetic directly to ros2 jazzy. I'd like to avoid having to do foxy first, then jazzy if we can avoid it.

clalancette, 2024-11-07:
For what it is worth, I think it will be *much* easier overall to attempt to port to Humble on Ubuntu 20.04.  Humble is supported as a "from-source" target on Ubuntu 20.04, and is still supported until 2027.

In other words:
1. Install Ubuntu 20.04.
2.  Install Noetic from packages as you normally would, following https://wiki.ros.org/noetic/Installation/Ubuntu
3. Follow the instructions to build Humble from source on Ubuntu 20.04: https://docs.ros.org/en/humble/Installation/Alternatives/Ubuntu-Development-Setup.html

(once you have ported to Humble, the jump to Jazzy should be much easier)

BlueNoise, 2024-11-07:
Does Humble support ros1 bridge? I only see the last rosbridge versions to be foxy and galactic.

sachinkum0009, 2024-11-08:
There is docker implementation for using ros1 bridge on humble https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder