ros2/ros1_bridge #460: What is my build setup missing?

andanders, 2025-07-03:
I've followed the README as well as I could, but my custom message is not being bridged from ROS2 to ROS1.

Everything builds wihout errors and `ros1_bridge` is bridging an unrelated ROS2->ROS1 topic that uses a standard message type, so it works.

Custom message package `inxpect_interfaces` in ROS1 and ROS2 has the same name and layout. 

### Environment

Using Ubuntu 20.04 (L4T/Linux4Tegra on Nvidia Jetson), Noetic and Humble.

ROS1 dist:	` /opt/ros/noetic/setup.bash`
ROS1 ws: 	` /home/nvidia/noetic_ws/src/inxpect_interfaces`
ROS2 Humble compiled from source    ` /home/nvidia/bitbucket/ros2_humble/install/setup.bash`
ROS2 ws:	 `/home/nvidia/humble_ws/src/inxpect_interfaces`

### env vars
```
ROS2_INSTALL_PATH=/home/nvidia/bitbucket/ros2_humble/install
ROS1_INSTALL_PATH=/opt/ros/noetic
```

### ros1_bridge package.xml
Added the following
```
<build_depend>inxpect_interfaces</build_depend>
<exec_depend>inxpect_interfaces</exec_depend>
```

### Checks
🆗 ROS1 package built
`rospack list-names | grep inxpect` returns `inxpect_interfaces`

🆗 ROS1 package built
`ros2 pkg list | grep inxpect` returns `inxpect_interfaces`

❌ sensor_field message not in ros1_bridge pairs
`. humble_ws/install/setup.bash `
`ros2 run ros1_bridge dynamic_bridge --print-pairs | grep field` returns nothing

closed 2025-07-04 by andanders

Danendra10, 2025-08-07:
how did you solved this one?

andanders, 2025-08-07:
Hi Danendra10. I never solved it because issue #445 blocks my progress.